opencpn-radar-pi/radar_pi #112: PPI windows not refreshed on Mac

douwefokkema, 2019-07-01:
HugoGr reported on the forum that the radar PPI windows are frozen on Mac with Mojave. I just verified this, is fully reproducible. It seems that the Render() is not called on Mac.
I checked older commits of radar_pi, problem remains, though these versions have been tested on Mac. I assume that recent changes in O5.0 are causing this issue. I am using the released and distributed O version 5.0.0 + 9065270.

douwefokkema, 2019-07-02:
There are 2 problems.
First is in radar_pi, beta2, not in released version. Solution has been pushed to master.
Second problem is in OpenCPN. The released version differs from the ones I compile myself. Released version does not function, my compiled versions are fine. Does Travis use different components than I use?

keesverruijt, 2019-07-21:
Sorry closed the wrong issue

keesverruijt, 2019-07-21:
This is fixed with plugin 5.0.1 or 5.0.3-beta2 and OpenCPN 5.0.1, see http://www.cruisersforum.com/forums/f134/opencpn-5-0-1-for-macos-mojave-10-14-5-a-220988.html

closed 2019-07-21 by keesverruijt