opencpn-radar-pi/radar_pi #116: Adding Furuno Radars to Radar Pi

sailing-yacht-asia, 2019-08-02:
Hi.

I would like to look at adding Furuno radar support to OpenCPN Radar Pi.

I would welcome some assistance with analysing Wireshark Traces from someone who has done this before for the other radar types supported - which should speed up the entire process.

Transfer between a Furuno Black Box radar and a plotter / MFD is UDP over Ethernet.

trudK45, 2019-08-02:
Hello!
I have done the reversed engineering for the Garmin xHD radar. It's a time consuming process to do this so you need to be prepared to be patient and stubborn.
I am willing to give you assistance at a level I am able to. I.e. a couple of hours a week.
/Hans

douwefokkema, 2019-08-02:
I would suggest you take a step by step approach with minimal changes per step. So connect the radar and MFD to the network, both standby. PC with wireshark connected. Get some exercise in using the filters on wireshark.
1. do nothing else but taking a wireshark recording. Analise it to find who is who. Locate IP addresses and ports. Who is sending what, are the messages UDP or TCP/IP.
2. Switch radar to ON from MFD. Analise the wireshark recording. What command is send to the radar to switch it on? How does the radar respond? 
3. Change some variable a single step. What are the changes in the radar response?
And so on. Be creative! 
Typical communication: MFD sends commands and stay alive messages, radar sends image data and status messages, normally on different ports and/or IP addresses. But ports and IP addresses may not be fixed, radar may advertise what ports and addresses it will use when asked for.

Douwe Fokkema

sailing-yacht-asia, 2019-08-04:
Thanks for the responses. Will revert once if I need some help or suggestions.

sailing-yacht-asia, 2019-08-04:
Hi

 

Thanks for your offer and response.

 

I am making reasonable progress at the moment and will get back to once I need some help or suggestions.

 

Paul

sailing-yacht-asia, 2019-08-04:
Hi

 

Thanks for this.

 

I am making reasonable progress at the moment and will get back to you once I need some help or suggestions.

 

Paul

bborchardt8, 2019-11-11:
Hi sailing-yacht-asia.
Have you progressed with work on Furuno radar? I am also interested in getting this to work and happy to put some time in.

jordanbigel-discovery, 2021-01-11:
@sailing-yacht-asia - hey now. Nothing new posted since Aug 19 leads me to believe you have either given up on doing this, or completed the work but have not been able to share it with the community. Can you give us any idea about what happened with this? 

@bborchardt8 -your comment was over a year ago - can I assume you have also not done this? 

I am dying to get my new Furuno working with OCPN but I don't want to spend weeks of time duplicating effort which someone else has already put it. 

Has anyone gotten the radar plugin to work with a Furuno NXT broadband radar yet?

THANKS! 
Jordan.

bborchardt8, 2021-01-11:
Hi, tried for some time but fell short in required skills - gave up.

jordanbigel-discovery, 2021-01-11:
Thanks for the reply. So far I have only managed to get the app to build on my machine. Long way to go.

bborchardt8, 2021-01-11:
Already done better than me…

Hakansv, 2021-01-11:
@bborchardt8 
The very first step would be to be familiar with Wireshark. Start your radar to work with the standard MFD or app. See if you can connect Wireshark to the same network and monitor the traffic. A filter to view only the (UDP?) radar traffic is handy.
Next is to sample events. Study what others have done for the same purpose and different examples here: https://github.com/opencpn-radar-pi/radar_pi/tree/master/example.
With proper pcaps from Wireshark it may be Douwe, Hans or Kees have time and joy to setup the plugin. It's also a good thing to try yourself. Study the code and different radar's network connection and data exchange.
Setup of the radar plugin built environment is described [here](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_windows) You have to first setup and build OCPN and then radar_pi. OR - Clone only radar_pi (git clone github.com/opencpn-radar-pi) and get you a opencpn.lib file. I have it if you need.

Hakansv, 2021-01-11:
BTW: I find Visual Studio 2017 to be good IDE and build motor on Win10.

NaatuMik, 2022-03-11:
I have a system with two Furuno Radars: DRS6A and DRS12A currently running with Windows PC Timezero Navigator. Now that Timezero announced they now longer supporting Navionics chart updates I would like to switch over to OpenCPN. I have been good programmer in the past many years ago, but my skills are now rusted. However I would be happy to contribute in this by testing or running some diagnostics or system discovery, for example.

bborchardt8, 2022-03-12:
Correct, I was unable to progress. 

Cheers
Bruce

zigzag930, 2022-12-11:
I have used Wireshark many times in the past and can provide any information required to assist with adding the Furuno DRS capability to the radar plug-in.    I already have the Garmin plugin radar plugin working.

As I am now adding a second radar (Furuno) to the boat, having the openCPN plugin working with the Furuno system would be awesome.

Let me know what I can do to assist.

keesverruijt, 2022-12-11:
Hi @zigzag930, you would need to start with cloning an existing radar into a new 'furuno' directory and then adapting the code.

To decide which radar to start with I would look at how the radar is detected and how it communicates with the radar. 
The Navico code is the most advanced; it listens to multicast addresses to determine the radar's IP addresses as these are "flexible"; even the port numbers change between radar types. Comms is generally multicast. 
The Garmin radars are simpler; a single fixed IP in a fixed IP range (172.16/12) must be chosen for the computer.

So try to find out how the Furuno works, then choose what fits the best, copy that code, rename everything from brand X to Furuno, then adapt the code.

antoine0183, 2022-12-22:
Hello @zigzag930, do you have any packet captures you can share already? I would like to start looking into the protocol but will only get my hands on a Furuno some time next quarter.

Hakansv, 2022-12-22:
I would probably have mentioned the /example folder in this repository. There are some pcap examples from other system if of interest.

keesverruijt, 2022-12-22:
Feel free to send pull requests to add these to examples, preferably with a little note in the form of a markdown file that explains what can be found in the file (the existing files do not have this, and it is a pain!)

Hakansv, 2022-12-22:
I know. It was a long time ago and I don't remember more than the file name says. That's way it was not uploaded before.

antoine0183, 2022-12-24:
> I've some very short attempts to cache some Furuno wifi traffic using Wireshark. Download [here](https://www.dropbox.com/s/8otctksfa8evtpn/Furuno_wifi.zip?dl=1)

Thank you for these. Unfortunately they are missing the control channel, since those packets were probably sent to the Radar unit directly and not over broadcast, but the pcaps are still a nice starting point.

zigzag930, 2022-12-24:
I intend to install my DSR25a over the period between Christmas and New Years.  Once done, I can help with packet traces, etc.  

In my case, the DSR25A uses a PSU013 power supply.   The “turn on” signal for power up is (to the best of my knowledge) not sent via regular packets and is done using an unused pair of wires in the RJ45 connector between the MFD and the PSU.   The power down sequence IS sent packets.     All other data except for the “power sync” startup signal are to be via regular traceable packets.

Jim

keesverruijt, 2022-12-28:
Guys, make sure that you don't get confused if something for one radar doesn't work with another. In our experience the different radars from the same brand don't have exactly the same protocol and messages that they use. Do not assume that that works for yours works for all of them, so make sure you name .pcap files with your (exact) radar model etc.

closed 2026-03-27 by keesverruijt