opencpn-radar-pi/radar_pi #117: Crash when unloading plugin

leamas, 2019-08-12:
When working with the new plugin loader, it turns out that  radar_pi crashes when unloaded. The code triggering crash (in src/PluginHandler.cpp at https://github.com/OpenCPN/OpenCPN/pull/1390):

    g_pi_manager->ClosePlugInPanel(pic, wxID_OK);
    g_pi_manager->UnLoadPlugIn(ix);
    string path = fileListPath(plugin_name);
    if (!ocpn::exists(path)) {
        wxLogWarning("Cannot find installation data for %s (%s)",
                     plugin_name.c_str(), path);
        return false;
    }
    ifstream files(path);
    while (!files.eof()) {
        char line[256];
        files.getline(line, sizeof(line));
        if (isRegularFile(line)) {
            int r = remove(line);
            if (r != 0) {
                wxLogWarning("Cannot remove file %s: %s", line, strerror(r));
            }
        }
    }
    wxLogMessage("Uninstall: complete.");
    return true;

Note the final log message.  Top of stacktrace:

    #0  0x00007ffff76445da in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /lib64/libwx_gtk2u_core-3.0.so.0
    #1  0x00007ffff76445e0 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /lib64/libwx_gtk2u_core-3.0.so.0
    #2  0x00007ffff74e4d03 in wxFrame::SendIdleEvents(wxIdleEvent&) () at /lib64/libwx_gtk2u_core-3.0.so.0
    #3  0x00007ffff7525e8d in wxAppBase::ProcessIdle() () at /lib64/libwx_gtk2u_core-3.0.so.0
    #4  0x00007ffff7451f15 in wxApp::DoIdle() () at /lib64/libwx_gtk2u_core-3.0.so.0
    #5  0x00007ffff7452017 in wxapp_idle_callback () at /lib64/libwx_gtk2u_core-3.0.so.0
    #6  0x00007ffff6bdd97b in  () at /lib64/libglib-2.0.so.0
    #7  0x00007ffff6be106d in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
    #8  0x00007ffff6be1438 in  () at /lib64/libglib-2.0.so.0
    #9  0x00007ffff6be1762 in g_main_loop_run () at /lib64/libglib-2.0.so.0
    #10 0x00007ffff64fcf4f in gtk_main () at /lib64/libgtk-x11-2.0.so.0
    #11 0x00007ffff746fc15 in wxGUIEventLoop::DoRun() () at /lib64/libwx_gtk2u_core-3.0.so.0
    #12 0x00007ffff6feee21 in wxEventLoopBase::Run() () at /lib64/libwx_baseu-3.0.so.0
    #13 0x00007ffff74d9ba6 in wxDialog::ShowModal() () at /lib64/libwx_gtk2u_core-3.0.so.0
    #14 0x0000000000638e46 in MyFrame::DoOptionsDialog() (this=0x10de7e0) at /home/mk/src/OpenCPN/src/chart1.cpp:6066

Final parts of opencpn.log:

    21:27:00 CEST: radar_pi: DeInit of plugin
    21:27:00 CEST: radar_pi: Emulator requested receive thread to stop
    21:27:00 CEST: radar_pi: Emulator receive thread stopped in 0 ms
    21:27:00 CEST: radar_pi: Radar A GuardZone 1: destroyed
    21:27:00 CEST: radar_pi: Radar A GuardZone 2: destroyed
    21:27:00 CEST: radar_pi: DeInit of plugin done
    21:27:00 CEST: Uninstall: complete.
    21:27:00 CEST: radar_pi: Emulator received stop instruction
    21:27:00 CEST: radar_pi: Emulator receive thread stopping

My understanding is that after DeInit() has completed, some idle processing continues which then references code no longer in memory. Part of the problem is that unloading this way is a new usecase.

leamas, 2019-08-13:
The crash is on Linux (fedora). Running on macos there is no crash. However, the log shows
    02:26:40 PDT: radar_pi: DeInit of plugin
    02:26:40 PDT: radar_pi: Garmin HD requested receive thread to stop
    02:26:40 PDT: radar_pi: Garmin HD receive thread stopped in 0 ms
    02:26:40 PDT: radar_pi: Radar A GuardZone 1: destroyed
    02:26:40 PDT: radar_pi: Radar A GuardZone 2: destroyed
    02:26:40 PDT: radar_pi: DeInit of plugin done
    02:26:40 PDT: radar_pi: Garmin HD received stop instruction
    02:26:40 PDT: radar_pi: Garmin HD receive thread stopping

That is, the receive thread is still active after DeInit() is completed, and that there is no crash is just a coincidence.

leamas, 2019-08-14:
On windows, also crash with log:

     12:06:26: radar_pi: DeInit of plugin
     12:06:26: radar_pi: Garmin HD requested receive thread to stop
     12:06:26: radar_pi: Garmin HD receive thread stopped in 0 ms
     12:06:26: radar_pi: Radar A GuardZone 1: destroyed
     12:06:26: radar_pi: Radar A GuardZone 2: destroyed
     12:06:26: radar_pi: DeInit of plugin done
     12:06:26: radar_pi: Garmin HD received stop instruction
     12:06:26: radar_pi: Garmin HD receive thread stopping

leamas, 2019-08-14:
The culprit seems to be the wxThread::Wait() function. The plugin `DeInit()` method invokes a shutdown on each RadarInfo instance. `Shutdown()` looks like:

     wxLongLong threadStartWait = wxGetUTCTimeMillis();
     m_receive->Shutdown();
     m_receive->Wait();
     wxLongLong threadEndWait = wxGetUTCTimeMillis();
     LOG_INFO(wxT("radar_pi: %s receive thread stopped in %llu ms"), m_name.c_str(), threadEndWait - threadStartWait);

m_receive->Shutdown basically sends a message the receiver thread  on a socket. Eventually, this message is picked up, resulting in the final log messages. However, at this point DeInit() is already completed.

The problem seems to be that wxThread::Wait() does not wait for the thread to complete, a wxWidgets bug. The log message ordering shows that this is indeed the case. It also shows that there is no actual delay when waiting.

leamas, 2019-08-14:
I have been poking around with the code, to no avail. This beats me: there are two "normal" ways to handle parallell things: threads and asynchronous IO based on select() or poll(). However, this code runs asynchronous IO in a thread... Hopefully, there is a trivial patch for this, but I just don't get it.

leamas, 2019-08-15:
It's hard to leave something like this... After cherrypicking a modified log from https://github.com/OpenCPN/OpenCPN/pull/1232 I get the following.: 

    22:43:30.727 MESSAGE pluginmanager.cpp:830 PlugInManager: Deactivating PlugIn: /home/mk/.local/lib/opencpn/libradar_pi.so
    22:43:30.727 MESSAGE radar_pi.cpp:370 radar_pi: DeInit of plugin
    22:43:30.727 MESSAGE GarminHDReceive.cpp:587 radar_pi: Garmin HD requested receive thread to stop
    22:43:30.728 MESSAGE RadarInfo.cpp:152 radar_pi: Garmin HD receive thread stopped in 1 ms
    22:43:30.734 MESSAGE GuardZone.h:106 radar_pi: Radar A GuardZone 1: destroyed
    22:43:30.734 MESSAGE GuardZone.h:106 radar_pi: Radar A GuardZone 2: destroyed
    22:43:30.735 MESSAGE radar_pi.cpp:409 radar_pi: DeInit of plugin done

This seems to imply that the only problem with the logging is that it comes unordered and without ms timestamps revealing this fact...  (merging opencp #1232  certainly feels like a good idea).

No idea why plugin crashes after unload.  Googling seems to indicate that stack or heap might be corrupted.

leamas, 2019-08-15:
With new log we have a more interesting stacktrace: 

    #0  0x00007ffff5dff81b in __strrchr_avx2 () at /lib64/libc.so.6
    #1  0x00007ffff5d2bff2 in basename () at /lib64/libc.so.6
    #2  0x000000000074b7b8 in OcpnLog::DoLogRecord(unsigned long, wxString const&, wxLogRecordInfo const&)
    (this=0xf345e0, level=3, msg=..., info=...) at /home/mk/src/OpenCPN/src/logger.cpp:103
    #3  0x00007ffff704928a in wxLog::CallDoLogNow(unsigned long, wxString const&, wxLogRecordInfo const&) ()
    at /lib64/libwx_baseu-3.0.so.0
    #4  0x00007ffff7049cdc in wxLog::FlushThreadMessages() () at /lib64/libwx_baseu-3.0.so.0
    #5  0x00007ffff7049e91 in wxLog::FlushActive() () at /lib64/libwx_baseu-3.0.so.0
    #6  0x00007ffff6fb0ddd in wxAppConsoleBase::ProcessIdle() () at /lib64/libwx_baseu-3.0.so.0

A breakpoint shows that this is when trying to log *radar_pi: Garmin HD received stop instruction*, with dangling pointers in ```wxLogRecordInfo& info```

This is effectively some flushing done as idle processing, after completing the unloading. At this point the thread is deleted....

keesverruijt, 2019-08-17:
@leamas we've seen crashes with logging before, and so far the recommended way forward was to not enable logging, but that is obviously not a real or even practical solution.

On the other hand, looking at the WX 3.1.2 source code, the only pointers are `filename`, `func`, `line` and `component`, how could any of those be invalid when the thread is deleted (but the DLL stays in memory, I presume...)

keesverruijt, 2019-08-17:
If the logging is really the issue, then maybe we should do something like this:

```wxLongLong threadStartWait = wxGetUTCTimeMillis();
 m_receive->Shutdown();
 m_receive->Wait();
 wxLog::FlushActive(); 
 wxLongLong threadEndWait = wxGetUTCTimeMillis();
 LOG_INFO(wxT("radar_pi: %s receive thread stopped in %llu ms"), m_name.c_str(), threadEndWait - threadStartWait);
```

which presumes the `DeInit()` is called on the main thread. In would work by doing a log flush before the thread object m_receive is deleted.

keesverruijt, 2019-08-17:
@leamas some quick testing shows that this doesn't ruin anything for me using the stock OpenCPN, can you affirm whether this fixes your issue? It certainly cleans up the ordering of the log, so I'm going to do this anyhow. 

I think that a better overall fix would be to do logging in all OpenCPN without this buffering which is -IMO- the wrong thing when logging to a file, but that is something that your https://github.com/OpenCPN/OpenCPN/pull/1232 should address?

leamas, 2019-08-17:
Thanks for fresh eyes and remarks!

As for the first code snippet,  this is more or less what wxWidgets does. I found a note somewhere that it flushes all thread logging once the thread has completed. Giving it a try, same crash:

    #0  0x00007ffff76445da in wxWindowBase::SendIdleEvents(wxIdleEvent&) (this=<optimized out>, event=...)
        at ../src/common/wincmn.cpp:2837
    #1  0x00007ffff76445e0 in wxWindowBase::SendIdleEvents(wxIdleEvent&) (this=this@entry=0x10def50, event=...)
        at ../src/common/wincmn.cpp:2837
    #2  0x00007ffff74e4d03 in wxFrame::SendIdleEvents(wxIdleEvent&) (this=0x10def50, event=...) at ../src/gtk/frame.cpp:182
    #3  0x00007ffff7525e8d in wxAppBase::ProcessIdle() (this=<optimized out>) at ../include/wx/list.h:1195
    #4  0x00007ffff7451f15 in wxApp::DoIdle() (this=0xe9b3b0) at ../src/gtk/app.cpp:159
    #5  0x00007ffff7452017 in wxapp_idle_callback(gpointer) () at ../include/wx/app.h:439
    #6  0x00007ffff6bdd97b in g_idle_dispatch
        (source=0x454b540, callback=0x7ffff7452000 <wxapp_idle_callback(gpointer)>, user_data=0x0) at gmain.c:5620
    #7  0x00007ffff6be106d in g_main_dispatch (context=0xed2b60) at gmain.c:3182
    #8  0x00007ffff6be106d in g_main_context_dispatch (context=context@entry=0xed2b60) at gmain.c:3847
    #9  0x00007ffff6be1438 in g_main_context_iterate
        (context=0xed2b60, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3920
    #10 0x00007ffff6be1762 in g_main_loop_run (loop=0x5780e20) at gmain.c:4116
    #11 0x00007ffff64fcf4f in IA__gtk_main () at gtkmain.c:1270
    #12 0x00007ffff746fc15 in wxGUIEventLoop::DoRun() (this=0x56b3e00) at ../src/gtk/evtloop.cpp:65
    #13 0x00007ffff6feee21 in wxEventLoopBase::Run() (this=this@entry=0x56b3e00) at ../src/common/evtloopcmn.cpp:78
    #14 0x00007ffff74d9ba6 in wxDialog::ShowModal() (this=0x493a3e0) at ../src/gtk/dialog.cpp:179
    #15 0x0000000000639f3c in MyFrame::DoOptionsDialog() (this=0x10def50) at /home/mk/src/OpenCPN/src/chart1.cpp:6066
    #16 0x0000000000631fa6 in MyFrame::DoSettings() (this=0x10def50) at /home/mk/src/OpenCPN/src/chart1.cpp:4975
    #17 0x0000000000630b36 in MyFrame::OnToolLeftClick(wxCommandEvent&) (this=0x10def50, event=...)
        at /home/mk/src/OpenCPN/src/chart1.cpp:458

When the crash happens the stack seems corrupted, the child is shit.

> but that is something that your OpenCPN/OpenCPN#1232 should address?

Not really. We need to stay compatible with the old logging statements -- that is, user the same framework. I have basically introduced a new backend + new frontend macros, all in the same overall machine.

keesverruijt, 2019-08-17:
> As for the first code snippet, this is more or less what wxWidgets does. I found a note somewhere that it flushes all thread logging once the thread has completed.
Well it doesn't, as adding that static method does change the order of the logging.

I'm not all that convinced that the issue is in the logging though. My personal hunch is that it has something to do with the extra PPI windows that the plugin creates, which are AUI windows. AUI is the worst abandonware in wxWidgets.

Does it help if you stop O, remove the AUI section in the ini file, then restart?

I'm about to do a long offshore passage, and am on board, so I don't have the (internet + time) means to pull your O version and start extended debugging sessions, sorry. Maybe valgrind gives a little better hint (although I found that valgrind makes O excruciatingly slow, to the point that it no longer is feasible to make real comparisons.)

leamas, 2019-08-18:
> It certainly cleans up the ordering of the log, so I'm going to do this anyhow.

Note that the proper way to do it is ```wxLog::GetActiveTarget()->Flush```, so the possibly overridden implementation is used.

> I'm not all that convinced that the issue is in the logging though.

Neither am I. It crashes even when all logging in thread is commented out.

> Does it help if you stop O, remove the AUI section in the ini file, then restart?

Will check... 

> I'm about to do a long offshore passage, and am on board, so

Happy sailing!  And thanks for input! 

For me, I need to make sure that this is indeed about the radar plugin. Need to test other plugins for this.

leamas, 2019-08-18:
> It certainly cleans up the ordering of the log, so I'm going to do this anyhow.

Note that the proper way to do it is ```wxLog::GetActiveTarget()->Flush```, so the possibly overridden implementation is used.

> I'm not all that convinced that the issue is in the logging though.

Neither am I. It crashes even when all logging in thread is commented out.

> Does it help if you stop O, remove the AUI section in the ini file, then restart?

Nope, same crash

> I'm about to do a long offshore passage, and am on board, so

Happy sailing!  And thanks for input!

leamas, 2019-08-18:
It's definitely the plugin code. If it isn't activated it can be removed without any crash.  

Strange thing is that if it *has* been activated it crashes, even if deactivated before uninistalling.

douwefokkema, 2019-10-04:
I compiled and installed your plugin branch on Windows10. Radar works fine. No crash on shutdown, I tried several times, with and without PPI windows. Below last part of the log. 

12:08:50: radar_pi: Navico 4G A received stop instruction
12:08:50: radar_pi: Navico 4G A receive thread stopping
12:08:50: radar_pi: Navico 4G A receive thread stopped in 0 ms
12:08:50: radar_pi: Navico 4G A destroy OpenGL canvas
12:08:50: radar_pi: Navico 4G B requested receive thread to stop
12:08:50: radar_pi: select maxFd=1360 r=1 elapsed=16
12:08:50: radar_pi: Navico 4G B received stop instruction
12:08:50: radar_pi: Navico 4G B receive thread stopping
12:08:50: radar_pi: Navico 4G B receive thread stopped in 0 ms
12:08:50: radar_pi: Navico 4G B destroy OpenGL canvas
12:08:50: radar_pi: Radar A GuardZone 1: destroyed
12:08:50: radar_pi: Radar A GuardZone 2: destroyed
12:08:50: radar_pi: Radar B GuardZone 1: destroyed
12:08:50: radar_pi: Radar B GuardZone 2: destroyed
12:08:51: radar_pi: DeInit of plugin done
12:08:51: PlugInManager: Deactivating PlugIn: D:\Program Files1\OpenCPNleamas\plugins\ShipDriver_pi.dll
12:08:51: opencpn::MyFrame exiting cleanly.
12:08:51: Chart cache purge
12:08:51: Chart cache PlugIn purge
12:08:51: PlugInManager: Deactivating PlugIn: D:\Program Files1\OpenCPNleamas\plugins\chartdldr_pi.dll
12:08:51: PlugInManager: Deactivating PlugIn: D:\Program Files1\OpenCPNleamas\plugins\radar_pi.dll
12:08:51: Closing NMEA Datastream Serial:COM5
12:08:51: Stopping Secondary Thread
12:08:53: Stopped in 2 sec.
12:08:51: radar_pi: NavicoLocate thread stopping
12:08:53: Chart cache purge
12:08:53: opencpn::MyApp starting exit.
12:08:53: LOGBOOK:  2019-10-04 10:08:53 UTC OFF: Lat   50.49850 Lon  -31.33450
12:08:53: UnLoading Tide/Current data source: D:\Program Files1\OpenCPN 4.99.0\tcdata\HARMONIC.IDX
12:08:53: opencpn::MyApp exiting cleanly...

leamas, 2019-10-04:
Yes [ thanks for testing!]  But what about uninstalling it? Still no crash?

douwefokkema, 2019-10-04:
No problem enabling / disabling the radar_pi. But in the "Install new plugins" window, radar-pi only has a dimmed "Update" button, no Uninstall button.

Douwe

leamas, 2019-10-04:
That is (knock, knock...) because you have installed the plugin as a traditional package (the installer treats such packages as read-only objects). To trigger the crash you need to remove the package and first install the plugin using the new installer page, possibly activate it and then uninstall.

douwefokkema, 2019-10-04:
I understand. Yes, crashing now on Uninstall. I do not think it is radar_pi all threads seem to end as they should, I will further investigate next week, off for the weekend.

leamas, 2019-10-04:
OK. One datapoint expanding on above: if you install the plugin as disabled (possibly requiring editing opencn.ini/conf) it is ok to uninstall it without crash. OTOH, if it has been activated at any point, uninstall crashes.

douwefokkema, 2019-10-09:
That is good information. Problem was indeed in radar-pi. I have pushed a solution to master. Message box was not deleted in de DeInit(). So I have added "delete m_pMessageBox;"
This commit is not yet tagged, I will make it a new beta after your testing.

leamas, 2019-10-09:
Just in case you wonder: I have rebuilt the binaries used by the new plugin loader, so it might be that you cannot reproduce the Uninstall crash any more. Will do some testing either late today or tomorrow (busy days...)

douwefokkema, 2019-10-09:
No, the crash can be not reproduced any more with the new version of radar_pi master, as I have pushed a solution to radar_pi master.

leamas, 2019-10-09:
And I have merged it into the binaries downloaded by the installer (they are built and deployed from my fork as long as #120 isn't merged).

As I understand it you have been able to reproduce the crash. If you cannot reproduce it now I see no problems tagging current master.

douwefokkema, 2019-10-09:
v5.0.4-beta4 including the solution mentioned above has been tagged on master.

leamas, 2019-10-09:
Fine with me. Thanks for your efforts!

Closing.

leamas, 2019-10-09:
I forgot to send some thanks to @keesverruijt ? Doing it here. Your remarks was really helpful at a time I was somewhat depressed over this seemingly simple bug.

closed 2019-10-09 by leamas

keesverruijt, 2019-10-09:
Thanks!

I must say that the more I use it the more I hate C++, just look at the fix for this bug.
In C you do all allocations manually, so it's easier to remember what is happening.
In Java/other modern languages, this wouldn't cause a crash at all.