opencpn-radar-pi/radar_pi #118: Garmin radar bearing alignment does not work for angles < 0

keesverruijt, 2019-08-17:
From http://www.cruisersforum.com/forums/f134/radar-plugin-v5-0-0-released-216057-11.html#post2948447

> Running OCPN 5.0.0 with radar_pi plugin v5.0.3-beta3 hooked up to a Garmin xhd2 scanner I run into the following problem (see attachment). When trying to adjust the bearing alignment it works well if the values are > 0 degrees. When doing a negative adjustment the value is recognized (see log attachement) but the value jumps back to 0 degrees immediately.
> Any ideas what the reason for this behavior could be.

and later after some tips:

> UPDATE - changing GarminxHDControl.cpp and comment line 228->230 seems to fix the issue. 

```
case CT_BEARING_ALIGNMENT: {
//if (value < 0) {
// value += 360;
//}

```

trudK45, 2019-08-17:
Are we sure the Garmin xHD behaves the same way as the Garmin xHD2 ? I am not back on my boat until beginning of November so I can't check this right now but from what I can remember the antenna alignment worked for both positive and negative alignments for my Garmin xHD radar with the previous code.

keesverruijt, 2019-08-17:
hmmm that is unfortunate if we had to differentiate between xHD and xHD2...

Leaving this open for now for others to chime in.

keesverruijt, 2019-08-17:
And of course, if @trudK45 can verify that the new code works with xHD as well in November then we can also close definitively.

octant-stenhh, 2019-08-18:
I will try to get my hands on a xHD scanner in addition to my xHD2 scanner to investigate if the two scanner types identify differently.

trudK45, 2020-03-02:
Unfortunately I had no time to check this when I was back on the boat in the US in November. Du to unexpected  events I had to return to Sweden after only 1 week in the water, The boat is now again on the hard in the US and I will not be back before November this year. Sorry folks.

trudK45, 2020-09-02:
This was tested by @iv3wjr and works for the Garmin xHD.

closed 2020-09-02 by trudK45

jussikoskinen, 2020-10-21:
This issue is also happening with Garmin HD radars (OpenCPN 5.2 with Radar plugin 5.1.4). Can you incorporate the change also the HD radars, please?

commit 1f9f3748 (2019-08-17): Fix #118: Remove suggested lines