opencpn-radar-pi/radar_pi #121: GarminHD Cannot change Range, Sea clutter.

SamHoulder, 2019-09-18:
Report/Log
http://www.cruisersforum.com/forums/f134/radar-plugin-v5-0-0-released-216057-14.html#post2964910

Workaround/Fix for range:
Change range definitions.
http://www.cruisersforum.com/forums/f134/radar-plugin-v5-0-0-released-216057-16.html#post2977718

Hakansv, 2019-09-28:
@bdbcat 
Dave..
Would you mind have a look at this observation of SamH? You're the G HD expertise here.
Thanks
Håkan

SamHoulder, 2019-09-28:
I gave this a little bit more thought and from what I figured out, the issue lays in that radar_pi use exact range_meter, where garminHD uses range_meter -1. That means "fixing" the garmin range definitions was a ugly hack, it was whats broke the nautical viewing in the control-window. 

I have since reversed the rangedefinitions back to what it was, and instead adjusted the range_meter where it is beeing sent/received. That is range_meters -1  when changing/sending range, range_meter+1 when receiving the message.  With that it works like a charm and displays correct.

For  seaclutter there is something strange going on. In radar_pi it can show som-ething like Status calm 100% when Im changing it from my 3010 plotter. I havent been able to compare the actual packets yet, my onboard switch wasnt to happy with ettercap/arppoisening, Ill have to tap in to the network cable to catch packets. More results for that in a couple of weeks when Im back.

keesverruijt, 2020-02-17:
I'm closing this as we implemented the range fixes in PR #123

closed 2020-02-17 by keesverruijt