opencpn-radar-pi/radar_pi #122: No overlay after  "True Heading" lost

lifgren, 2019-10-03:
**Problem: No radar overlay.**

what happens:
"True heading" and "Magnetic heading" check mark disappears 10 secs after starting, preventing radar overlay. Non-overlay radar works/oriented correctly.

version: radar_pi-5.0-ov50-win32

radar preferences: 
checked: pass radar heading to OpenCPN (no overlay either setting) 

radar window:
initially checked: OpenGL, position True/Magnetic heading, Variation.
Then 10 secs after starting, heading checks both disappear.

setup:
connected to RI10 interface is: point1 GPS antenna, Lawrence(Navico) 3G, windows PC 
connected to windows PC: usb GPS and usb heading sensor for HDT

nmea debug window shows GPHDT does not stop:
<GREEN>11:21:41 (Serial:COM5) $GPHDT,356.7,M*f<0x0D><0x0A>
<GREEN>11:21:41 (Serial:COM5) $GPHDT,356.8,M*0<0x0D><0x0A>
<GREEN>11:21:41 (Serial:COM5) $GPHDT,356.8,M*0<0x0D><0x0A>
<GREEN>11:21:41 (Serial:COM5) $GPHDT,356.8,M*0<0x0D><0x0A>
<GREEN>11:21:41 (Serial:COM5) $GPHDT,356.7,M*f<0x0D><0x0A>
<GREEN>11:21:41 (Serial:COM5) $GPHDT,356.5,M*d<0x0D><0x0A>
<GREEN>11:21:41 (Serial:COM5) $GPHDT,356.5,M*d<0x0D><0x0A>
<GREEN>11:21:41 (Serial:COM5) $GPHDT,356.5,M*d<0x0D><0x0A>
<GREEN>11:21:41 (Serial:COM6) $GPGGA,152141.779,,,,,0,00,,,M,0.0,M,,0000*5D<0x0D><0x0A>
<GREEN>11:21:41 (Serial:COM5) $GPHDT,356.5,M*d<0x0D><0x0A>
<GREEN>11:21:41 (Serial:COM6) $GPGSA,M,1,,,,,,,,,,,,,,,*12<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM5) $GPHDT,356.6,M*e<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM6) $GPRMC,152141.779,V,,,,,,,031019,,,N*4C<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM5) $GPHDT,356.6,M*e<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM5) $GPHDT,356.5,M*d<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM5) $GPHDT,356.6,M*e<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM5) $GPHDT,356.8,M*0<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM5) $GPHDT,356.9,M*1<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM5) $GPHDT,356.8,M*0<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM5) $GPHDT,356.9,M*1<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM5) $GPHDT,357.0,M*9<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM6) $GPGGA,152142.779,,,,,0,00,,,M,0.0,M,,0000*5E<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM5) $GPHDT,357.1,M*8<0x0D><0x0A>
<GREEN>11:21:42 (Serial:COM6) $GPGSA,M,1,,,,,,,,,,,,,,,*12<0x0D><0x0A>

douwefokkema, 2019-10-03:
Lifgren, 
Can you set in the opencpn.ini file VerboseLog=1. Delete the opencpn.log, and run your system again such that the problem occurs. After that attach the new opencpn.log here.
Douwe Fokkema

lifgren, 2019-10-03:
please see attached verbose log file using the radar emulator

I'll get to the boat this evening and send a non-emulator log

Same issue using radar powered up, emulator or radar powered down: heading
'checks' disappears and no map overlay

Thank you in advance

Hakansv, 2019-10-04:
@lifgren 
_"--radar preferences:
checked: pass radar heading to OpenCPN (no overlay either setting)--"_
If you have selected "Pass radar heading to O..." please deselect it. This is to pass a heading coming from the radar or the RI10. Your heading is obviously coming via serial/USB so not via the radar interface.
Håkan

douwefokkema, 2019-10-04:
Sorry, but I do not see a log file.

lifgren, 2019-10-04:
Hmmm, i see log file attached to the email.

Try link to my google drive:
 https://drive.google.com/file/d/1M6aHFoIeMevMNsAyf1AuBrWjnxxuBVYX/view

douwefokkema, 2019-10-05:
An emulator session is not using a heading. Such a log will not help to solve your issue.
Douwe

lifgren, 2019-10-07:
Using the Lowrance radar power up, please find:

1. verbose log of radar session 
 at https://drive.google.com/file/d/1tZnW8th74p1mIpNciLk8KTg2wrOqlKdM

2. screen shot at https://drive.google.com/file/d/1zV4m1clKpM9o1ccfxZ4nxjV80n5Mplym

In screenshot note:
- nmea log with accepted heading and GPS sentences yet radar window with no checks in heading boxes.
- radar working fine as no-overlay

douwefokkema, 2019-10-09:
Your nmea0183 messages are wrong. After the * there should be a 2 digit checksum.

closed 2019-10-15 by douwefokkema