opencpn-radar-pi/radar_pi #124: A new Halo24 does not pass the initialization stage. New ports or message?

Hakansv, 2019-10-22:
@keesverruijt  @douwefokkema 
A user, Tim, has a quit new Halo24. It's connected OK but not leaving initialization.
OpenGL, SOG, COG is there but it won't get ready for Start Transmit.
Due to lack of knowledge for the Halo I cannot figure out from the Wireshark capture what the code is waiting for.  Would you mind to have a look.

oepencpn.log, renamed config file and also renamed pcapng from the start up are attached.
Screen shot of the initialization phase also attached.

Håkan
[opencpn-1.log](https://github.com/opencpn-radar-pi/radar_pi/files/3755222/opencpn-1.log)
[opencpn.ini.txt](https://github.com/opencpn-radar-pi/radar_pi/files/3755229/opencpn.ini.txt)
[halo_radar.pcapng.txt](https://github.com/opencpn-radar-pi/radar_pi/files/3755232/halo_radar.pcapng.txt)

[image]

Karanga15, 2019-10-22:
I have since found v5.0.3 and installed that. It made no difference

Hakansv, 2019-10-23:
From a mail from Tim:

This morning, I tried again.
But this time, instead of trying to connect to Halo A, I tried to connect to Navico BR24, Navico 3G, Navico 4G A and 4G B.
I managed to connect to BR24 and 3G!! 
But not the others.

Then, I unconnected from the ones working, and I can now connect to Halo A. Hallelujah!!! 
But not to Halo B.
Halo A appears to be working well. Even if Halo B is not working, if I try to disconnect from Halo B, opencpn crashes every time. 

I am attaching a screenshot of the Halo A info. This shows the firmware and date. I don't know how to upload this screenshot to github, that is why I am sending it to you. Perhaps you could upload it? 
This info window pops up quite often. I wonder if it does this when heading info is lost (I am resting at anchor, so there is no accurate or consistent boat speed or heading).

Thank you so much for your help so far. 
Tim.

Håkan>>:
Didn't succeed upload the pic. The firmware says it's from Nov 2 2018 so would be similar to Kees' Halo??

Hakansv, 2019-10-23:
@Karanga15 
Tim..
Without a compass heading using COG while anchored for overlay the result would be very unstable. Until you've a heading try to keep to the PPI window instead of chart overlay.
Håkan

douwefokkema, 2019-10-24:
Tim,
In the attached tim.ini (rename to opencpn.ini) I have manually entered the IP and port info for data, report and send for both Halo A and B into your original opencpn.ini. These data comes from your log file, there the radar addresses are listed for your radar. But due to a bug they did not get to the right location. For your Halo A the adresses happen to be the same as the 4G A, so there for you got the Halo A working after selecting 4G A. With the attached ini file both radars should work. At least if your radar does not switch to different addresses after a while. Please give it a try. In the meanwhile I will look for a update in radar_pi, will take some time.

Douwe 

(I anchored same spot near Low Islands 10 years ago, great!, see http://percussion.nl/de_wereldreis/naardetop.html  )

[tim.ini.txt](https://github.com/opencpn-radar-pi/radar_pi/files/3766744/tim.ini.txt)

Karanga15, 2019-10-24:
Thank you so very much for for doing this for me. I really appreciate the
stuff you guys do!
I just read some of your story. It was funny to read you talking about the
rolling swell at high tide - exactly what I am sitting in right now.
I will rename the file and put it in the correct directory.
Tim

douwefokkema, 2019-11-29:
This issue should be solved with v5.0.4-beta6. The previous version did not recognize the radar when replaying your pcapng file. The new version immediately picks up the radar. Please test with your radar as well. To make the test meaningful, delete the Radar0NavicoInfo and Radar1NavicoInfo from the opencpn.ini first.

Karanga15, 2019-11-29:
Hi Douwe
Thank you for the update, and thank you for letting me know about it.
I will install the updated version when I get back to the boat.
Hopefully this version will also fix the issue I discovered that the plugin
continually searches for the radar connection. If the radar is off (no
power), the log file reports no connection. I don't know how often, but
that report creates a HUGE log file. The biggest file so far is 55 Gb!
My pc began to crash, until I found out why. In testing, even when OpenCPN
is running for only 5 seconds, creates a log file of 31 Mb...
Anyway, as I said, hopefully this new version will resolve this minor
issue.
Kindest regards
Tim

Karanga15, 2019-11-30:
Ok, I have downloaded and installed the updated version. I assume it is the
correct version, but it does not have the 5.0.4 label...

I put " " marks either end of the navico0info and navico1info instructions
in the ini file. I didn't want to delete them just yet!

The new plugin version appears to have better functionality. Thank you!

However, I have noticed that it is more sensitive to clouds (I think). But
not at all range levels? Attached screenshots show this. The real clouds
out the window are very light and not dense and I don't think should be
displayed at all. I have tried to adjust gain and rain clutter, but if I
adjust enough to remove the clouds, I don't get displays of real boat
targets.
Please note that I am still learning how to optimise radar settings!

Also, maybe the plugin has features not compatible with my Halo24, but I
find some settings don't seem able to be modified. For instance the auto
Sea clutter harbour/off-shore/weather setting can be changed but always
changes itself back to default auto harbour after about 5 seconds.
Similarly, target boost always changes itself back to Off.

Tim

Karanga15, 2019-11-30:
Update...
The clouds that I thought were showing on the radar have now gone from the
sky. There is only water and blue sky to the east of my position.
However, at those same range levels, I continue to get a large area of
apparent radar return.
This did not happen with the previous version of radar-pi that I was
running.
Tim

douwefokkema, 2019-12-01:
Hi Tim,
OpenCPN will still have the 5.0.0 label, but the radar_pi plugin should be v5.0.4-beta6, this is shown in the radar info window.
The main reason for the update was to solve the detection problem of the radar. I understand that that is solved on your system now.
I did not change anything on the radar image itself (as far as I know). But I cannot look at your issue as I do not see the attached screenshot that you speak about. If you mail it to me I will have a look.
Regarding the too big logfile with the radar off: make sure that in your opencpn.ini VerboseLog=0, that will limit logging. If that does not help, please send me some of the repeating lines that fill up your log.

Already in up in Darwin? We were first anchored in Fanny bay, long way out for the tide. Later in Tipperary marina, very comfortable.

Cheers, Douwe

Karanga15, 2020-01-08:
Hi Douwe
Just a small bug in my system.
After sailing for a few hours, the radar menu seems to switch all options
to "noise rejection - off" as the only one available. However, clicking on
an option does go to the correct sub-menu.
After a reboot of O5 and the plugin, the menu option becomes correct again
(until a few hours later).
I have attached a screenshot. I hope you can view this.
Tim

douwefokkema, 2020-02-13:
The connection problem for the Halo is solved with the beta 6 version.

closed 2020-02-13 by douwefokkema