opencpn-radar-pi/radar_pi #132: Radar selection causes a crash in RPi and more Linux.

Hakansv, 2020-04-14:
Compare: https://www.cruisersforum.com/forums/f134/radar-plugin-v5-0-0-released-216057-25.html#post3116747
Using radar_pi v 5.1.1. My next step is to bisect down in commits to try to find when this was introduced. 
Kees, Douwe...  Any direct thoughts? 
Bug Description:
When selecting a radar type and click OK O hangs or crash. 

I can confirm the crash in both RPi3 and RPi4 even if built on each machine.
I can also confirm it's the radar selection method causing the bug.
@chrisryan43. A work around for the very moment, thanks to LeaseOnLife, is to close OCPN and edit your /home/pi/.opencpn/opencpn.conf file.
Change the setting: [PlugIns/Radar] Radar0Type to read exact like this:
Radar0Type=Navico BR24
Save the file and start OpenCPN again. That's working for me on my RPi3.

Hakansv, 2020-04-14:
If I built and checkout   v5.0.4-beta2, 13c228fba, on my RPi3 the radar selection works wo crash or hang. So something introduced after that. I'll check more tomorrow. Now late.

LeaseOnLife, 2020-04-14:
Since happy hour is still a few minutes away in Pacific Mexico, I played a bit more with Hakan's results: beta6 still crashes for me, beta4 seems fine, tested on Kubuntu 20.04, not touching the PI4 for now...

LeaseOnLife, 2020-04-14:
more: the crash seems to be caused by [https://github.com/opencpn-radar-pi/radar_pi/commit/a21e4e6b8289f59b7de4687bee2b3ad52de3d926](url). Test performed: System running with Navico 4A and 4B. deselected these two radars, and selected emulator. played with emulator for a moment. Went beck to select 4A and 4B => crash

keesverruijt, 2020-04-14:
That looks like a likely candidate, and it's about the only complicated stuff changed lately.

Assigning to Douwe for now. I will take over if he can't find it.

Hakansv, 2020-04-15:
The crash is in this method:
radar_pi : 502
On my test r=0
`for (r = 0; r < M_SETTINGS.radar_count; r++) {
      if (m_radar[r] && m_radar[r]->m_radar_type != oldRadarType[r]) {
>>> Runs to here. Next statement "Shutdown" is causing the crash.
        m_radar[r]->Shutdown();
        RemoveCanvasContextMenuItem(m_context_menu_control_id[r]);
        delete m_radar[r];
        CLEAR_RADAR_INFO;
        m_radar[r] = new RadarInfo(this, r);
      }`
So, why crash on RPi but not on Win?

douwefokkema, 2020-04-17:
Should be solved with commit 1e876fe in master.

LeaseOnLife, 2020-04-17:
almost.... I think normal use-cases will work, it does for my Navico 4A/B deselection/selection. I found a case where I still can make it crash: 
Select "Emulator" only as radar. apply, close all dialogs. Then go back into the radar selection => crash. After restarting O I can select my Navicos again !?!

douwefokkema, 2020-04-17:
Thanks for testing. I will investigate further over the weekend.

Hakansv, 2020-04-17:
LeaseOnlife. Good catch. It's the same on my RPi3.

Hakansv, 2020-04-17:
And, again it's for me only on Linux. It's not reproducible in Win. (Have no MacOs)

douwefokkema, 2020-04-18:
I just pushed another update to master. Solves the described issue on my system.

LeaseOnLife, 2020-04-18:
that seems to do the trick. thanks!

Hakansv, 2020-04-18:
Thanks Douwe.
Linux is sometimes a strange system? Cares a lot about trivia. :)
Impressive how you found that one.
Now it works fine on Win and Linux; RPi4 and Rpi3

douwefokkema, 2020-04-18:
In both cases the crashes were on an assignment operation a = b, with a and b structures of type NavicoRadarInfo. And that structure contains a wxString. It seems to me that windows copies the whole structure regardless of the contents where Linux (the compiler) uses wxWidgets to copy the wxString. And crashes when no wxString is found.

closed 2020-04-18 by douwefokkema