opencpn-radar-pi/radar_pi #135: maintain rainge settings between O restarts

LeaseOnLife, 2020-05-07:
as the title says... on my 4GA and 4GB radar settings fro range are being set to "auto" after O restart. I think this setting should 'stick'

keesverruijt, 2020-05-25:
We'd have to be careful not to force a radar range onto the radar when there are MFDs or multiple OpenCPN instances sharing the radar. You don't want an OpenCPN start to mess with someone else's carefully selected ranges.

If you leave the radar on, it will remember the setting.

So to do this, we'd need an extra control option "save radar ranges on exit" and we already have so many.

LeaseOnLife, 2020-06-06:
understood! Maybe a way to save and load preferred settings (harbor, coastal, offshore, etc) would be a way?