opencpn-radar-pi/radar_pi #136: OpenCPN 5.1.511 crashes on exit if radar_pi activated.

douwefokkema, 2020-05-12:
last log line: 18:43:15.767 MESSAGE NavicoReceive.cpp:639 radar_pi: Navico 4G B receive thread stopping. Radar_pi is 5.1.3

I reverted to O 5.1.428, no difference.

Hakansv, 2020-05-12:
On O 5.1.511
I've tried to end O while transmitting.
I could repeat something similar only once with the Emulator. Last log entry: "Deactivating PlugIn: C:\Users\admin\AppData\Local\opencpn\plugins\radar_pi.dll"
The next five time O ended normally.
I've also tried the BR24 single and together with Emulator a couple of times. Couldn't reproduce. All times O ended normally.
Håkan

douwefokkema, 2020-05-13:
Håkan,
You are right and so am I. Crash only occurs with oeSENC plugin enabled.
Douwe

douwefokkema, 2020-05-14:
Earlier this crash was fully reproducible, today only on one machine in 50% of the cases. Strange. I attached Visual Studio debugger. Crash in iosfwd, see screenshot.
[image]

keesverruijt, 2020-05-25:
Sorry guys, I can't debug OpenCPN on Windows until I get home. Next time can you show the Call Stack?

Hakansv, 2020-05-25:
Win10
Using present O beta, oeSENC and radar_pi I can't reproduce this issue.  
I'm using 3G. Is it a difference for 4G A/B?
Håkan

douwefokkema, 2020-05-27:
Solved in f2c660. Problem was in NavicoLocate, I added a Wait() after m_locator->Shutdown.

closed 2020-05-27 by douwefokkema