opencpn-radar-pi/radar_pi #137: Having multiple services receiving the same Radar data

Hjorth2, 2020-05-18:
Hi

Would it be possible to have a program running simultaneously with the Radar_pi, that receives the same data from the Radar?
For context, I'm in a university project that works with route planning. As such we have made a program that is able to record the data and do some computations on it. However, we would like to have a visualization running with overlay on the sea charts at the same time as recording the data. Preferably, we would have the Radar_pi controlling the Radar settings and visualizing the data, while having a different program simply receiving the same data and do some computations with it.
Would this be possible without changing the code of the Radar_pi?
Thank you.

douwefokkema, 2020-05-18:
As the radar data is UDP any application can receive the same data simultaneously. For example you could run wireshark and record all or part of the data. If you also put the navigational data out to the net in a UDP stream, you can even replay all and get the same view as when recording.

Hjorth2, 2020-05-25:
Thank you. Using your answer and modifying the C code found [here](https://gist.github.com/hostilefork/f7cae3dc33e7416f2dd25a402857b6c6) I was able to extract the UDP data.

closed 2020-05-25 by Hjorth2