opencpn-radar-pi/radar_pi #140: Understanding Navico ranges

Hjorth2, 2020-06-04:
Hi,
I was hoping you could clarify the ranges for me. 
I'm using a 3G Lowrance Radar. When setting the range of the Radar plugin I notice that it still shows objects that are further away than that range. According to the information given by Lowrance the max range of this Radar, should be 24NM, but even then there are objects seen further away.

I have been trying to look at the radar_pi code and the data received from the Radar. When the range is set to the max of 24NM, it looks like the actual range is 72704m. Does this mean that the actual max range of the Radar is 72km? 
At the same time when the range is set to the minimum of 50m it looks like that the actual range is 98m.

Thank you for your time.

douwefokkema, 2020-06-04:
That is the way Navico works. In radar_pi we handle ranges the same way as the Navico displays. And you are right, this is strange. We request a certain range from the radar. Then the radar adjusts its range and tells us what the new range is. And the new range is about (differs from time to time) 50% larger than requested. When the range is set to 24 NM, the actual range could very well be 72704 m (I did not verify). But this does not mean that the radar will see targets at that distance, returns from targets that far away will normally be to weak to be detected. To better see targets outside the range, you can drag the image in the radar window off-center.

Hjorth2, 2020-06-04:
I see. So when the Radar returns the measurements in 512 bytes per spoke, the last byte corresponds to the actual range (e.g.72704 m) and not the requested range (24 NM)?

douwefokkema, 2020-06-06:
Yes.