opencpn-radar-pi/radar_pi #141: Radar stayalive message should be sent when O is in background

keesverruijt, 2020-06-21:
According to a user, and at least on Windows with O5, the RenderGLOverlay call is not made when the program is hidden or maybe even when it is just not visible because it is hidden behind other windows.

We should call the StayAlive code in a different way.

douwefokkema, 2020-06-21:
I just checked. When another window has focus and is on top of O with radar, both the overlay and the radar windows are refreshed as usual. I did not check with O completely in the background.

Hakansv, 2020-06-21:
On O Beta 5.1.620 and recent radar_pi.
I can't either reproduce. The radar transmit is alive when O is in background of other programs and also when minimized, i.e. not viewed on the desktop.

douwefokkema, 2020-08-02:
I checked with my radar on board. Yes, when you close the radar window and have no overlay radar will time out and got to standby. Solution is straightforward, UpdateTransmitState() is the only one doing StayAlive. So do the following and all is fine. Tested on board. I can make a commit for this one of the next weeks, have some others in the pipeline that need further testing first. Or anyone else can apply the change to master.

   }
 
-  if (!m_pi->IsRadarOnScreen(m_radar)) {
-    return;
-  }
-
   if (state != RADAR_OFF && TIMED_OUT(now, m_stayalive_timeout)) {
     m_control->RadarStayAlive();

douwefokkema, 2020-08-02:
In the above, read the dots as minus signs!

douwefokkema, 2020-08-30:
The original issue can not be reproduced, when the radar is visible on screen and O is hidden, the radar is refreshed as normal. But if the radar is not visible on screen (no overlay, no PPI windows) it is not refreshed, independent of O is on screen or not. The solution given solves not the issue as described originally (can not be reproduced), but the other issue. So now the radar will be refreshed also if it is not visible in OpenCPN. See commit 285918, tested and merged in master.

keesverruijt, 2021-03-12:
This is still reported as being a problem with 5.2.0-beta4.

douwefokkema, 2021-03-12:
The correction that I applied in 929b2c6 (see above) solves this issue for Windows 10, tested. It seems that the issue still exists on Unix.

Hakansv, 2021-03-12:
I today did a test on board using 3G and in opposite to my earlier comments this bug is present also in Win10. A minimized OCPN and radar_pi is not updating the "stay alive" message so the scanner goes to Stand by

douwefokkema, 2021-03-17:
Things are much worse. It is not only the StayAlive that does not work with a minimized OpenCPN, guard zones are not checked and ARPA targets are not refreshed. With a minimized O, RenderGLOverlayMultiCanvas() is not called any longer. Better don't minimize O.

douwefokkema, 2021-03-22:
Update of StayAlive, Guard zones, ARPA removed from RenderGLOverlayMultiCanvas and now in separate TimedUpdate thread, running every 500 msec.
    The new thread is independent of the display process and also runs when O is minimized. This solution involves no changes in the different radar specific modules, the receive and locate threads are not modified. See 0d8251f in master. Requires further testing.

LeaseOnLife, 2021-03-22:
I would be happy to test.... but even after a fresh git clone, radar_pi is asking (and not finding) winsock2.h on my linux system. 
see https://github.com/opencpn-radar-pi/radar_pi/issues/161

Hakansv, 2021-03-24:
Douwe
Tested positive on Win10 and RPi4, Buster using BR24.
Both accept to minimize wo stay alive interruptions. 
I have though not tested for more than half an hour of transmit time on each device. This note with respect to the thread clash discussion. I'll do a longer time test on the boat next visit.

keesverruijt, 2021-04-11:
Fixed with commit 0d8251f1990603aa9507e548f5ca5b34ad0e2c4d

closed 2021-04-11 by keesverruijt