opencpn-radar-pi/radar_pi #143: Empty radar charts on Jetson AGX (aarch64)

kizilkayabayram, 2020-07-17:
Radar_pi charts loads empty. I've reinstalled Ubuntu bionic, OpenCPN and radar_pi but the issue still persists. However on my x86 machine everything works fine. Here are my kernel info, screenshot and logs. I've couldn't come up with anything, so I'll appreciate any little comment. Thanks. 

`4.9.140-tegra #1 SMP PREEMPT Thu Jun 25 21:22:12 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux`

Screenshot: 

[image]

 Logs:

`14:26:19.117    INFO logger.cpp:169 ------- OpenCPN version 5.2.0 restarted at 2020-07-17 -------

14:26:19.117    INFO logger.cpp:169 Using loglevel TRACE
14:26:19.117 MESSAGE chart1.cpp:1883 wxWidgets version: wxWidgets 3.0.4 Linux 64 bit wxGTK
14:26:19.133 MESSAGE chart1.cpp:1888 OS Version reports as:  4.9
14:26:19.133 MESSAGE chart1.cpp:1890 MemoryStatus:  mem_total: 15814 mb,  mem_initial: 321 mb
14:26:19.133 MESSAGE chart1.cpp:1899 SData_Locn is /usr/share/opencpn/
14:26:19.133 MESSAGE chart1.cpp:1915 PrivateDataDir is /home/agx3/.opencpn
14:26:19.133 MESSAGE chart1.cpp:1972 Using existing Config_File: /home/agx3/.opencpn/opencpn.conf
14:26:19.149 MESSAGE navutil.cpp:1198 winPluginDir, read from ini file: 
14:26:19.149 MESSAGE navutil.cpp:1291 Setting Ownship Lat/Lon 33.358, -79.282
14:26:19.152 MESSAGE navutil.cpp:3160 Switch InlandEcdis mode Off
14:26:19.160 MESSAGE styles.cpp:854 Styles loading from /usr/share/opencpn/uidata/styles.xml
14:26:19.163 MESSAGE styles.cpp:831 No styles found at: /home/agx3/
14:26:19.163 MESSAGE styles.cpp:831 No styles found at: /home/agx3/.opencpn/
14:26:19.180 MESSAGE styles.cpp:786 Using SVG Icons
14:26:19.180 MESSAGE chart1.cpp:2030 Detected display size (horizontal): 508 mm
14:26:19.181 MESSAGE OCPNPlatform.cpp:925 Getting DefaultSystemLocale...
14:26:19.181 MESSAGE chart1.cpp:2067 System default Language:  en_US
14:26:19.181 MESSAGE chart1.cpp:2070 Config file language:  en_US
14:26:19.181 MESSAGE chart1.cpp:2075 Adjusted App language:  en_US
14:26:19.181 MESSAGE OCPNPlatform.cpp:1010 ChangeLocale: Language load for:  en_US
14:26:19.183 MESSAGE OCPNPlatform.cpp:1062 ChangeLocale: Locale Init OK for:  en_US
14:26:19.183 MESSAGE OCPNPlatform.cpp:1081 Loading catalog for opencpn core.
14:26:19.184 MESSAGE OCPNPlatform.cpp:925 Getting DefaultSystemLocale...
14:26:19.184 MESSAGE chart1.cpp:2083 Opencpn language set to:  en_US
14:26:19.184 MESSAGE ConfigMgr.cpp:686 Loading Configs catalog: /home/agx3/.opencpn/Configs/configs.xml
14:26:19.188 MESSAGE chart1.cpp:2305 Creating MyFrame...size(1298, 1056)  position(622, 24)
14:26:19.190 MESSAGE chart1.cpp:2877  **** Connect stuff
14:26:19.471 MESSAGE chcanv.cpp:2350 Metrics:  m_display_size_mm: 508     g_Platform->getDisplaySize():  1920:1080   
14:26:19.471 MESSAGE chcanv.cpp:2355 wxDisplaySize(): 1920 1080
14:26:19.471 MESSAGE TCDataSource.cpp:52 Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONIC.IDX
14:26:19.675 MESSAGE chartdb.cpp:240 ChartDB Cache policy:  Application target is 1024 MBytes
14:26:19.675 MESSAGE chart1.cpp:2473 GPS Watchdog Timeout is: 6 sec.
14:26:19.681 MESSAGE chart1.cpp:2564 OpenCPN Initialized in 1163 ms.
14:26:20.198 MESSAGE chart1.cpp:6860 OnInitTimer...0
14:26:20.353 MESSAGE routeman.cpp:1251 DPMM: 3.77953   ScaleFactorExp: 1
14:26:20.353 MESSAGE routeman.cpp:1258 Looking for UserIcons at /home/agx3/.opencpn/UserIcons/
14:26:20.353 MESSAGE navutil.cpp:1612 Loading navobjects from navobj.xml
14:26:20.353 MESSAGE navutil.cpp:1623 Done loading navobjects, 0 duplicate waypoints ignored
14:26:20.355 MESSAGE gshhs.cpp:1377 Loading World Chart Q=0 in 0 ms.
14:26:20.355 MESSAGE gshhs.cpp:179 Background world map loaded from GSHHS datafiles found in: /usr/share/opencpn/gshhs/
14:26:20.454 MESSAGE chart1.cpp:6860 OnInitTimer...1
14:26:20.571 MESSAGE chart1.cpp:6860 OnInitTimer...2
14:26:20.571 MESSAGE pluginmanager.cpp:1069 PlugInManager: loading plugins from /home/agx3/.local/lib/opencpn;/usr/local/lib/opencpn;/usr/lib/opencpn;/usr/lib/opencpn
14:26:20.571 MESSAGE pluginmanager.cpp:1022 Using LD_LIBRARY_PATH: /home/agx3/.local/lib/opencpn:/usr/local/lib/opencpn:/usr/lib/opencpn:/usr/lib/opencpn:/usr/local/cuda-10.2/lib64:
14:26:20.571 MESSAGE pluginmanager.cpp:1058 Using PATH: /home/agx3/.local/bin:/usr/local/bin:/usr/bin:/usr/bin:/usr/local/cuda-10.2/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
14:26:20.571 MESSAGE pluginmanager.cpp:1075 Loading plugins from dir: /home/agx3/.local/lib/opencpn
14:26:20.572 MESSAGE pluginmanager.cpp:1101 PlugInManager searching for PlugIns in location /home/agx3/.local/lib/opencpn
14:26:20.572 MESSAGE pluginmanager.cpp:1116    Directory /home/agx3/.local/lib/opencpn does not exist.
14:26:20.572 MESSAGE pluginmanager.cpp:1075 Loading plugins from dir: /usr/local/lib/opencpn
14:26:20.572 MESSAGE pluginmanager.cpp:1101 PlugInManager searching for PlugIns in location /usr/local/lib/opencpn
14:26:20.572 MESSAGE pluginmanager.cpp:1116    Directory /usr/local/lib/opencpn does not exist.
14:26:20.572 MESSAGE pluginmanager.cpp:1075 Loading plugins from dir: /usr/lib/opencpn
14:26:20.572 MESSAGE pluginmanager.cpp:1101 PlugInManager searching for PlugIns in location /usr/lib/opencpn
14:26:20.573 MESSAGE pluginmanager.cpp:1138 Found 5 candidates
14:26:20.573 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libchartdldr_pi.so
14:26:20.576 MESSAGE pluginmanager.cpp:2098 Plugin is compatible: true
14:26:20.576 MESSAGE pluginmanager.cpp:2161 PlugInManager: Loading PlugIn: /usr/lib/opencpn/libchartdldr_pi.so
14:26:20.582    INFO pluginmanager.cpp:2332 PlugInManager:  /usr/lib/opencpn/libchartdldr_pi.so
[... 49 lines cut ...]
14:26:20.970 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libdashboard_pi.so
14:26:20.970 MESSAGE pluginmanager.cpp:1193 Skipping not enabled candidate.
14:26:20.970 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libradar_pi.so
14:26:20.970 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libgrib_pi.so
14:26:20.970 MESSAGE pluginmanager.cpp:1193 Skipping not enabled candidate.
14:26:20.971 MESSAGE OCPNPlatform.cpp:1010 ChangeLocale: Language load for:  en_US
14:26:20.971 MESSAGE OCPNPlatform.cpp:1062 ChangeLocale: Locale Init OK for:  en_US
14:26:20.972 MESSAGE OCPNPlatform.cpp:1075 Loading catalog for:  opencpn-chartdldr_pi
14:26:20.972 MESSAGE OCPNPlatform.cpp:1075 Loading catalog for:  opencpn-wmm_pi
14:26:20.972 MESSAGE OCPNPlatform.cpp:1075 Loading catalog for:  opencpn-radar_pi
14:26:20.973 MESSAGE OCPNPlatform.cpp:1081 Loading catalog for opencpn core.
14:26:20.973 MESSAGE OCPNPlatform.cpp:925 Getting DefaultSystemLocale...
14:26:21.134 MESSAGE chart1.cpp:6860 OnInitTimer...3
14:26:21.144 MESSAGE NavicoLocate.cpp:297 radar_pi: Located radar IP 192.168.1.17 port 1, interface 192.168.1.18 [1906Z62037/236.6.7.100:6132/236.6.7.102:6134/236.6.7.101:6133]
14:26:21.144 MESSAGE NavicoLocate.cpp:314 radar_pi: Located radar IP 192.168.1.17 port 2, interface 192.168.1.18 [1906Z62037/236.6.7.103:6135/236.6.7.105:6137/236.6.7.104:6136]
14:26:21.144 MESSAGE NavicoLocate.cpp:297 radar_pi: Located radar IP 192.168.1.17 port 1, interface 172.17.0.1 [1906Z62037/236.6.7.100:6132/236.6.7.102:6134/236.6.7.101:6133]
14:26:21.144 MESSAGE NavicoLocate.cpp:314 radar_pi: Located radar IP 192.168.1.17 port 2, interface 172.17.0.1 [1906Z62037/236.6.7.103:6135/236.6.7.105:6137/236.6.7.104:6136]
14:26:21.235 MESSAGE chart1.cpp:6860 OnInitTimer...4
14:26:21.872 MESSAGE options.cpp:5819 options: got device count: 1
14:26:22.041 MESSAGE chart1.cpp:7405    ***GPS Watchdog timeout at Lat:33.358   Lon: -79.282
14:26:22.089 MESSAGE chart1.cpp:6860 OnInitTimer...5
14:26:22.190 MESSAGE chart1.cpp:6860 OnInitTimer...6
14:26:22.190 MESSAGE chart1.cpp:7155 OnInitTimer...Last Call
14:26:22.191 MESSAGE chart1.cpp:7173 OnInitTimer...Finalize Canvases
14:26:25.064   DEBUG colourcmn.cpp:188 wxColour::Set - couldn't set to colour string 'inherit'`

douwefokkema, 2020-07-17:
I have the same problem from time to time on Windows 10. Same screen as you have. Can not find a solution, restarting OpenCPN often works, sometimes I needed to recompile. I have concluded that this should be an hardware issue and ordered new motherboard, processor and memory. We will see if this works.

kizilkayabayram, 2020-07-17:
> I have the same problem from time to time on Windows 10. Same screen as you have. Can not find a solution, restarting OpenCPN often works, sometimes I needed to recompile. I have concluded that this [...]

In my situation, sometimes control menu disappears. But I've never seen the charts working on ARM64. I've recompiled it several times, even the dependencies (wxWidgets ie.) but it's still the same.

douwefokkema, 2020-07-20:
I am doubting now if it is really a hardware issue in my case. Bayram, could you get me your opencpn.ini file, I would like to check it. 
Douwe

kizilkayabayram, 2020-07-21:
I also executed opencpn inside a clean docker container to see if the issue persists. However I've got the some results as the host machine. Charts successfully displayed on both x86 host and docker container running inside that host. And it failed to load charts inside docker container running on arm host. 

Here, I've attached my opencpn config file. 

[opencpn.txt](https://github.com/opencpn-radar-pi/radar_pi/files/4953549/opencpn.txt)

Thanks.

douwefokkema, 2020-07-25:
kizilkayabayram,
Could you try the following: delete in the opencpn.ini file the contents of the section headed by "[AUI]" until the next section starting with "["  . Then try to run OpenCPN as you did before. If it runs OK, it might only be a single time, O rewrites the section you have deleted. In my test case it did run fine several times (with the deleted section present again from the second time) until it went wrong again.

Second experiment: In the opencpn.ini file, find the section [PlugIns/Radar]. In that section you will find Radar0WindowShow=1 and Radar1WindowShow=1. Make both = 0. Your OpenCPN should run fine now, with radar windows hidden. You can expect a crash as soon as you try to show a radar window through menu Window -> Hide/Show PPI. Please try if this is the case.

Let me know the outcomes of both tests. These are not solutions, but a way to find the solution. 
Sorry this takes some time, it is a complex issue and not fully reproducible on my side, and I am on a sailing trip right now.
Douwe

douwefokkema, 2020-07-25:
And please make VerboseLog=14 in the ini file, this will generate some more output in the log, and let me have the log.

kizilkayabayram, 2020-07-27:
Alright, I'll try them and post the results here . Sorry I'm posting late too, I'm on a trip now and as soon as I return back I'll try all. Thanks .

douwefokkema, 2020-07-28:
kizilkayabayram,
Forget about the above. I think I found the issue and need a few days to prepare a new version (on a sailing trip now). Will let you know.
Douwe

douwefokkema, 2020-07-29:
I have made a patch that might bypass your issue, if your issue is the same as I was working on. You will find it at https://github.com/opencpn-radar-pi/radar_pi in branch test2. Please try.

Douwe

kizilkayabayram, 2020-07-30:
> I have made a patch that might bypass your issue, if your issue is the same as I was working on. You will find it at https://github.com/opencpn-radar-pi/radar_pi in branch test2. Please try.
> 
> Douwe

I'll be on my workstation in a couple of days. After trying I'll update this post.

douwefokkema, 2020-08-08:
It is a bit trial and error, please try branch issue_143. It solves a similar problem on my system and makes the system more robust anyway. Please reply in a separate message, updates don't generate mail, I missed your update above for some time.
Douwe

kizilkayabayram, 2020-08-13:
I've tried with new branch and played around a little bit, but nothing changes in my situation. I'll try to dig deeper to examine this issue. Here I'm dropping my config and logs.  Thanks.
[opencpn_conf.txt](https://github.com/opencpn-radar-pi/radar_pi/files/5068050/opencpn_conf.txt)
[opencpn_log.txt](https://github.com/opencpn-radar-pi/radar_pi/files/5068051/opencpn_log.txt)

douwefokkema, 2020-08-13:
Something wrong here. Your opencpn.ini shows verboselog = 14, the opencpn.log gives a loglevel of 0. That explains why there is so little logging. I think your system is reading the wrong opencpn.ini. It should be in the .opencpn directory. But I just made a version for testing with Raymarine radars. It has lots of updates all over and should work fine with Halo. It is still in my private github, needs some restructuring, has lots of logging and a fixed loglevel of 12, independent of the ini file setting. It is : https://github.com/douwefokkema/radar_pi/ branch raymarine. I suggest you try that one, we will get at least some logging.

kizilkayabayram, 2020-08-14:
I've noticed that too,I think I've opened log file directly from opencpn about menu, but I'm not sure. Anyway here is my new log file. 
[opencpn_log.txt](https://github.com/opencpn-radar-pi/radar_pi/files/5074356/opencpn_log.txt)

keesverruijt, 2020-08-16:
The logfile shows:

14:31:12.965 MESSAGE radar_pi.cpp:1207 radar_pi: RenderOverlay
14:31:12.973 MESSAGE NavicoReceive.cpp:515 radar_pi: select maxFd=32 r=0 elapsed=251
14:31:12.973 MESSAGE NavicoReceive.cpp:391 radar_pi: Navico Halo B no address to listen on
14:31:13.077 MESSAGE NavicoReceive.cpp:515 radar_pi: select maxFd=24 r=0 elapsed=250
14:31:13.077 MESSAGE NavicoReceive.cpp:391 radar_pi: Navico Halo A no address to listen on
14:31:13.155 MESSAGE NavicoReceive.cpp:515 radar_pi: select maxFd=32 r=0 elapsed=251

this means your Ethernetnet card doesn't have an IP address. This is not normal, even if there is no DHCP server on the wired ethernet it should be assigned an automatic IP address in the 192.168 range.

Does your Jetson even see the network?
If you do not have a DHCP server, run one on your Jetson tied to the Ethernet card, or assign a network address manually.

We need to fix our code so that it recognises the case that there are no cards to monitor as an explicit error and show the error dialog.

douwefokkema, 2020-08-16:
There is still another possibility in case your system does have an IP address. Halo radars start without report and data address. Or these are in the config file, or these come from the radar itself, picked up by the Locator. As long as there is no data or report address, nothing will be received and no interface will be selected. In your case I have the impression that the receive thread is using all the time, leaving no cycles for the Locate thread to do its job. Moreover, Receive has prio 100, Locate 50. And after a failure reading Receive will immediately read again, running in blind circles. 
So I did the following: Added after the "no address to listen on" a  wxMilliSleep(100), so that there will be some room for Locate to do its job. And I adjusted the priority of Locate upwards to 100 and Receive down to 90. 
Besides that I had some illegal symbols in the NavicoRadarInfo string. Initialization was wrong, corrected in Init() and in MakeRadarSelection().
You will find these changes in branch test143. 
Additionally we could consider to stop the Locator as soon as the radar has been found. Might save some performance.

kizilkayabayram, 2020-12-09:
Finally figured it out. Workaround is much simpler than I thought. Inside the default config file parameter "OpenGL" is set to "0". After changing it to  "1" charts showed up. It's set to "0" by default inside auto generated config file.

Tested with opencpn 5.2.4 and radar_pi v5.0.5-ov50 on Jetson Agx running Ubuntu 18.04.

keesverruijt, 2020-12-11:
That's weird, if OpenGL was 0 then you should be getting a plugin "status" dialog that shows that OpenGL condition is not satisfied. You can set this in the OpenCPN GUI settings dialog under "advanced", you don't need to edit the .ini file by hand.

closed 2020-12-13 by kizilkayabayram