opencpn-radar-pi/radar_pi #145: Garmin xHD no transmit sector does not work for angles < 0

iv3wjr, 2020-08-19:
Running radar plugin v5.14-ov51 dated 2020-05-24 with OpenCPN 5.2.0. and a Garmin RADAR xHD on a Raspberry pi 4 with raspbian buster kernel 5.4.51-v7l+.
My installation has a bearing alignment of +90 degrees (RADAR is looking starboard) but I could not set the no  transmit sector: for my installation I was able to set the end zone to -80 but could not set the start zone to -100: any start value less than 0 is not accepted. Positive values work well.

I suggest also to allow values in the range [0, 360] for any angle as the PPI indicator correctly shows this range on the screen.

trudK45, 2020-08-26:
@iv3wjr 
Now that you are able to run tcpdump, could you do a recording when sending these commands ?
I will then have a look at it and see if I can figure out what is going on.
Looking at the code it seems as if you should be able to set negative values on both angles.
How and if the fact you are pointing your radar dome to starboard does screw things up I don't know,
and if the alignment to the bow is accounted for or not when the radar dome sets start and stop angles.
However this you should be able to figure out by trying different angles for alignment and start stop angles if
you have reliable tx on/ff working.

iv3wjr, 2020-08-26:
@trudK45 
As soon as I will return to my beloved I will do the recording for you.
If I remember well the "no transmit sector" accepts **one** negative angle but not both. If you look at the opencpn.log I uploaded for the TX issue you will see that "no transmit zone end" was set to -80 and "no transmit start" to 150. With end set to -80 I could **not** set "start" to -110.  These angles seem referred to the bow, + to starboard. Visually I could see RADAR returns starting at -80 (i.e. 280° with bow at 0° on the PPI) and ending at +150°. Pressing the +10° key repeatedly reaches the maximum of 180. Pressing the -10° did not work either but I do not remember why, probably because end cannot be less than start (but here both angles are negative). This is why I suggested to use "marine" angles in range [0, 360], in my experience negative angles always bring weird problems. But a session log should be exhaustive.
 I have no compass so I look always to the PPI window.

trudK45, 2020-08-26:
@iv3wjr 
But what you see there is the values previously set in the radar dome's memory. These are values sent from the dome
and received in the radar_pi. You were not able to send a tx on command and I suspect no commands to the radar
could reach the radar dome so you were not able to change any settings. The values displayed on screen are values 
fetched from the radar., not the values you enter. So, if not sent they will never propagate back to the display.

iv3wjr, 2020-08-26:
@trudK45 
I know. Those were the last values I set when I could talk to the dome. I played with +10/-10 degrees for the "no tx zone" then I disabled the no transmit zone. From opencpn.log I uploaded two days ago:
...snip
14:19:57.240 MESSAGE GarminxHDReceive.cpp:621 radar_pi: Garmin xHD 0x0930: bearing alignment 90
14:19:57.260 MESSAGE GarminxHDReceive.cpp:696 radar_pi: Garmin xHD 0x093a: no transmit zone mode 0
14:19:57.270 MESSAGE GarminxHDReceive.cpp:707 radar_pi: Garmin xHD 0x0940: no transmit zone start 150
14:19:57.280 MESSAGE GarminxHDReceive.cpp:714 radar_pi: Garmin xHD 0x0941: no transmit zone end -80
...
so it was clear to me that the values were stored into the dome's memory.
In my previous post (for this issue) I was talking about a session (reconstructed looking at the logfile) when I was able to start/stop the Garmin xHD and play with all the settings. I hope this helps.

iv3wjr, 2020-08-28:
@trudK45 
In attachment opencpn.log and tcpdump.pcap files of a session where I tried to set the "no transmit sector" to the range [260, 280] (or [-100, -80] for using the radar_pi syntax).
At time 18:08:51.882 you will see:
no transmit zone start 134217718
this happened when I was decreasing the no transmit start value by -10 degrees as you can see a few lines above.
Have fun debugging
[opencpn20200827.log.gz](https://github.com/opencpn-radar-pi/radar_pi/files/5140620/opencpn20200827.log.gz)
[tcpdump20200827_0.pcap.gz](https://github.com/opencpn-radar-pi/radar_pi/files/5140621/tcpdump20200827_0.pcap.gz)

trudK45, 2020-08-28:
@keesverruijt 
I think we have a bug here.

In GarminxHDReceive.cpp:461 we have

typedef struct {
  uint32_t packet_type;
  uint32_t len1;
  uint32_t parm1;
} rad_ctl_pkt_12;

I.e. parm1 seems to be declared as unsigned but actually it is 2's compliment in this case.
The value is sent from the radar as FFFF FEC8  = -10 when right shifted 5 bits but written to the log file as 134217718 (7FFF FFF6).

If I compare with what is coded for Dome offset the line
GarminxHDReceive.cpp:622 looks like
        LOG_VERBOSE(wxT("radar_pi: Garmin xHD 0x0930: bearing alignment %d"), (**int32**_t)packet12->parm1 / 32);
Here we have a cast to int32 for parm1.

But for the no transmit start  there is no cast.

GarminxHDReceive.cpp:707 looks like
        LOG_VERBOSE(wxT("radar_pi: Garmin xHD 0x0940: no transmit zone start %d"), packet12->parm1 / 32);

and GarminxHDReceive.cpp:714 iooks like
        LOG_VERBOSE(wxT("radar_pi: Garmin xHD 0x0941: no transmit zone end %d"), (int32_t)packet12->parm1 / 32);
Here we also have a cast.

Obviously line 707 should have the cast to be able to handle negative numbers. Also line 709 should have it.

I think this mishap is due to us not anticipating you would want to set the no transmit start to port.

I don't have a development environment set up so it would take me a while to make a build. Are you, Kees, able to help out here or are you Roberto able to build ?

iv3wjr, 2020-08-28:
@trudK45 
Following the instructions on the github page to my great surprise I was able to build the deb package at the first attempt on my Kubuntu 20.04 machine at home. I will try to do the same on the Raspi on my Orion.
Once I have the package is it ok to install it with
dpkg -I radar_pi_5.0.5-ov50_amd64.deb ?
The OpenCPN-Options-Plugins dialog has an "Import plugin" button that likes tarball files so I think this is made for something else.
I will keep in touch as soon as I am able to build the plugin on the Raspi.

trudK45, 2020-08-28:
@Hakansv 

> Once I have the package is it ok to install it with
> dpkg -I radar_pi_5.0.5-ov50_amd64.deb ?

Håkan should be able to answer this one.

iv3wjr, 2020-08-28:
@trudK45 
Successfully built radar_pi_5.0.5-ov50_armhf.deb
Now I'm waiting for the cpp source file to replace.

trudK45, 2020-08-28:
Here you are.
I have introduced the casts on the two lines.
Try it out.

[GarminxHDReceive.cpp.gz](https://github.com/opencpn-radar-pi/radar_pi/files/5143528/GarminxHDReceive.cpp.gz)

Hakansv, 2020-08-28:
What I use to do on my RPi when I want to test the home brewed version is to:
- Uninstall the managed version from within the O pluginmanager. (If installed)
- Build locally with:  make && sudo make install.
- Then the pluginmanager will find no managed plugin and then search in the "old" location and load that. If I later will use a managed version I install that and it will be prioritized. 
Works for me

iv3wjr, 2020-08-29:
@trudK45 
@Hakansv 
Thanks for your efforts. I built the plugin following Hakansv intructions and now I am able to set correctly my no tx sector. I used [-93, -87] to point out that I have a blind  sector to port. In my opinion the two casts to int32_t solved the issue.

Regarding  the oter issue I opened a few days ago about starting the RADAR I suspect it depends on a weird setting of my eth0 on the Pi. I will ivestigate more about this and I will post there when I get close to a solution.

Really thank you guys for your time.
Roberto

trudK45, 2020-08-30:
I have done a pull request so Kees can merge in the fixes for this bug.

keesverruijt, 2020-08-30:
Merged.

closed 2020-08-30 by keesverruijt