opencpn-radar-pi/radar_pi #146: Garmin xHD: "transmit: turn on" sometimes does not work

iv3wjr, 2020-08-21:
Running radar plugin v5.14-ov51 dated 2020-05-24 with OpenCPN 5.2.0. and a Garmin RADAR xHD on a Raspberry pi 4 with raspbian buster kernel 5.4.51-v7l+.

I experienced a weird behaviour of the "Start/Stop radar" button: most of the times it works as it should, putting the RADAR in transmit mode and back to standby. Sometimes pressing the "Start/Stop radar" button does not start the transmission.

I activated logging with VerboseLog=1 and compared the two opencpn.log files.
When the RADAR starts I see:
8:12:04.806 MESSAGE GarminxHDControl.cpp:152 radar_pi: Garmin xHD transmit: turn on
18:12:04.815 MESSAGE GarminxHDReceive.cpp:573 radar_pi: Garmin xHD 0x0919: standby/transmit 1
18:12:04.826 MESSAGE GarminxHDReceive.cpp:536 radar_pi: Garmin xHD reports status STARTING
18:12:04.945 MESSAGE GarminxHDReceive.cpp:565 radar_pi: Garmin xHD 0x0916: scan speed 0
18:12:04.995 MESSAGE GarminxHDReceive.cpp:760 radar_pi: Garmin xHD 0x0993: state-change in 9830 ms

When the RADAR does **not** start I see for example:
11:16:16.863 MESSAGE GarminxHDControl.cpp:152 radar_pi: Garmin xHD transmit: turn on
11:16:16.905 MESSAGE GarminxHDReceive.cpp:577 radar_pi: Garmin xHD 0x091e: range 926
11:16:16.945 MESSAGE GarminxHDReceive.cpp:589 radar_pi: Garmin xHD 0x0924: autogain 2
11:16:16.955 MESSAGE GarminxHDReceive.cpp:594 radar_pi: Garmin xHD 0x0925: gain 4600
11:16:16.965 MESSAGE GarminxHDReceive.cpp:599 radar_pi: Garmin xHD 0x091d: auto-gain mode 0
11:16:16.965 MESSAGE GarminxHDReceive.cpp:615 radar_pi: Garmin xHD m_gain.Update(46, 1)
11:16:16.985 MESSAGE GarminxHDReceive.cpp:631 radar_pi: Garmin xHD 0x0933: rain mode 0
11:16:17.005 MESSAGE GarminxHDReceive.cpp:646 radar_pi: Garmin xHD 0x0934: rain clutter 4000
11:16:17.045 MESSAGE GarminxHDReceive.cpp:654 radar_pi: Garmin xHD 0x0939: sea mode 0
11:16:17.058 MESSAGE radar_pi.cpp:990 radar_pi: rendering took 0 ms, PPI0= 0 ms, PPI1= 1, Overlay0= 0, Overlay1= 37685760 next render in 250 ms

The file is populated only by
Garmin xHD 0x0919: standby/transmit 0
messages

Seems to me that the "turn on" command gets lost. When the RADAR did not start I tried to quit OpenCPN and restart it: same behaviour. I tried also:
- reboot the Raspi
- power off and restart the Raspi
- power off the RADAR and restart it

with no succes.

Three days later when I returned to my boat the RADAR started/stopped correctly and I could save the logfile. Both files are available for inspection.

iv3wjr, 2020-08-24:
Yesterday I tried again to start the RADAR plugin with no success.
Here are the logfile and the tcpdump file of the session:
[opencpn_20200823.log](https://github.com/opencpn-radar-pi/radar_pi/files/5116205/opencpn_20200823.log)
[tcpdump_20200823.dump.gz](https://github.com/opencpn-radar-pi/radar_pi/files/5116214/tcpdump_20200823.dump.gz)
I hope this can help to catch the issue.

trudK45, 2020-08-24:
@keesverruijt 
Is this normal ?
>14:19:38.581 MESSAGE pluginmanager.cpp:1232 PlugInManager: Radar has loaded very slowly: 154 ms

iv3wjr, 2020-08-24:
Seems that also oeSENC and Dashboard loaded very slowly. I don't know if this is normal, maybe it depends on the Raspi. Do you have other users running OpenCPN + radar_pi on the Raspi 4?

trudK45, 2020-08-24:
I think Håkan is running radar_pi on Raspi but I have not heard that he was having any problems.
I will analyze the tcpdump-file and get back when I have any result.

Hakansv, 2020-08-24:
I've a lot of these on both Win and RPi4. That's normal since the new plugin management in 5.2 I suppose. 
This is from my RPi4:
PlugInManager: Radar has loaded very slowly: 144 ms
There are similar entries for the other plugins. from 50 to 700 ms.
Håkan

trudK45, 2020-08-24:
@Hakansv 
Tahnks Håkan.

trudK45, 2020-08-24:
@iv3wjr 
Looking at your tcpdump-file I don't see any command sent to the radar. It looks like the status goes from WARMUP to STANDBY and then nothing else.
1) Does the display show WARMING_UP up and then STANDBY ?
2) Did you send a transmit command after the initial warm up period ?

iv3wjr, 2020-08-24:
I agree. I never used tcpdump so forgive me for the wrong extension I gave to the file. I installed Wireshark on my home pc and this morning I had a first look at the dump: no TX from the Raspi to the RADAR.
1. I confirm: I waited 60 seconds while WARMING_UP and as soon as STANDBY appeared
2. I clicked 5 times the Start button
You can find in attachment also the opencpn.log file of the same session where you can see my attempts to start the RADAR at 
11:25:41.210, 11:25:43.881, 11:25:44.677, 11:25:45.264 ...
seems that the message was emitted but the command was not.

keesverruijt, 2020-08-24:
Yes, don't worry about the slow plugins, Dave should have put the limit at 500ms not 50 or so!

trudK45, 2020-08-24:
@keesverruijt 
Thanks Kees.
I don't have a system running here so I can''t check myself.
Where does the output off wxLogError() go ?

keesverruijt, 2020-08-24:
I think the normal logfile.

trudK45, 2020-08-24:
@iv3wjr 
Would it be possible for you to set/change VerboseLog=5 in the opencpn.ini file ?
You would need to do that before invoking OpenCPN.
I would be interested to see if then the data package is written to the log file.
If it is then I think that is as far as I get without having a system to run on.

Just a check: Do you have a firewall running ?

iv3wjr, 2020-08-25:
As soon as I get on board I will increase log verbosity for a new test. I hope this evening.
No firewall running.
Thanks for the hint.

iv3wjr, 2020-08-26:
Yesterday I was able to start/stop RADAR transmission without problem. I set VerboseLog to 5 and I could see the following message:
MESSAGE GarminxHDControl.cpp:124 radar_pi: Garmin xHD Garmin xHD transmit 9 bytes:  19 09 00 00 01 00 00 00 01
that is exactly what I expected looking at the source code.
Same for  "transmit: turn off" with the last byte set to 00.
I saved both the opencpn.log and the tcpdump.pcap of the session but I think you don't need them as everything worked as it should.
I will repeat the tests in the next days, now I can only suspect a hardware issue on my Raspi. In my job career as a programmer (36 years) I found only *once* a faulty memory bank that drove me crazy, so never say never.
Again thank you for your time, now I have the tools to analyze what is happening. I will keep you informed.

trudK45, 2020-08-26:
@iv3wjr 
The only thing changed is the verboselog log level. At least that is what I suppose. If I where you
I would change it back to 1 and see what happens. If the error then returns it would probably suggest something is
wrong in a memory since I have heard of no one else having this problem. Only other thing I can think of is a
 intermittently  faulty ethernet cable. RX-pair is different from TX-pair.

trudK45, 2020-08-30:
@iv3wjr 
We have another issue "Garmin xHD radar bearing alignment does not work for angles < 0 #118" open.
Since I at the moment does not have access to my xHD radar I am not able to check if the change of the code
still work after it was changed by @keesverruijt  .
Are you able when you are in your boat check if the xHD radar still works for alignments to port ?
It would be much appreciated and if it still works then we could close this issue.

iv3wjr, 2020-08-30:
No problem, I hope to be on board tuesday or wednesday. About this issue I suppose I have a problem with the network interfaces: eth0 and wlan0. Network configuration for the Raspi has changed and probably I have some garbage left in the configuration files. Is it possible that I send correctly the "turn on" command but the packet is routed to the wrong (wlan0) interface?

iv3wjr, 2020-09-01:
@trudK45 
In attachment two screenshots, one with positive bearing alignment of 90 and one with negative bearing alignment of -90. 
[image]
[image]
Both are ok.
Now I can start/stop the RADAR without problem: you gave me the right hint  when you wrote "RX-pair is different from TX-pair" so I thought of a possible routing problem.
At the end this was not an issue but only a wrong configuration on my Raspi. Thanks again for your help.

trudK45, 2020-09-02:
As this seem to be ok, I close this issue.

closed 2020-09-02 by trudK45