opencpn-radar-pi/radar_pi #152: Garmin HD radar bearing alignment does not work for angles < 0

jussikoskinen, 2020-10-21:
Garmin HD radars have the same issue as reported for the Garmin xHD radars (fixed #118). As far as I know, the fix was only implemented for xHD radars. Can you please also implement it for HD radars.

Hakansv, 2020-10-21:
@jussikoskinen 
I don't know much about Garmin but if you're on Windows I made a quick patch for the same range code adjustment on HD as for xHD. >> https://www.dropbox.com/s/0m8i344cvkyryhu/radar_pi-5.0.51-ov50-win32.exe?dl=1
If you already have the managed version installed please uninstall, using OCPN Options->Plugins handle, and then install my package for a test. If you want to reinstall the official managed version again just do so in OCPN Plugins. (The managed version will be prioritized if installed.)

trudK45, 2020-10-21:
@Hakansv 
Removing the 3 lines around GarminHDControl.cpp: 242 

if (value < 0) {
 value += 360;
}

most probably solves the issue.

Hakansv, 2020-10-21:
@trudK45 
Yes, that's what I did for my test version.
Thanks
Håkan

jussikoskinen, 2020-11-29:
Sorry @Hakansv for the delayed reply. I am actually using Linux (Raspberry Pi) and the boat is also already in winter storage, so I am not able to test it at this point.

Hakansv, 2020-11-29:
@trudK45 
@bdbcat 
I could easily merge this patch but since I can't test a Garmin user comment would be valuable. Is it so a general Garmin unit admit a negative bearing alignment value, like -5 degr. ( Apart from Navico where we have to make it always positive, like 355 degr.)

trudK45, 2020-11-29:
@Hakansv 
@bdbcat 
Most probably the alignment for the Garmin is in the range +- 180 degrees.
However, since my boat is in the US and I am in Europe I have no way of verifying the code.
Håkan I suggest you go ahead and do the merge. It most probably is correct and if it is not we will have to take it from there.
/Hans

Hakansv, 2020-11-29:
@trudK45 
Hans..
Thanks for your comment. I do agree and will merge it soon.
Håkan

Hakansv, 2020-11-29:
Fixed by: 8382ee2e
@jussikoskinen 
I close this and hope for positive test next spring. If not please come back.
Thanks for the catch.

closed 2020-11-29 by Hakansv

jussikoskinen, 2021-05-27:
I can confirm that the fix fixed the issue. Thank you very much!