opencpn-radar-pi/radar_pi #153: flatpak metadata: wrong <target>

leamas, 2020-11-17:
The flatpak metadata contains `<target>flatpak</flatpak>`, should be `<target>flatpak-x86_64</flatpak>`. Effectively, this makes the plugin hidden and un-installable. See https://github.com/OpenCPN/OpenCPN/issues/2095

Hakansv, 2020-11-17:
@leamas 
As you may have seen in radar_pi that flag is present here:
elseif (OCPN_FLATPAK)
    set(PKG_TARGET "flatpak")
    set(PKG_TARGET_ARCH "x86_64")

But PKG_TARGET_ARCH may not be useful? I'm neither aware nor the originator of this so from my side it's just a curious question.

rgleason, 2020-11-17:
Hakan, it's confusing, but origin decisions have affected this. I think you need "flatpak-x86_64" in PKG_TARGET.

leamas, 2020-11-17:
The problem is that when you look into the metadata template here is

    <target>${PKG_TARGET}</target>

PKG_TARGET  expands to just flatpak, so it doesn't work.  Problem is that these variables are used in multiple places, not just in the metadata. So  a safe, proper fix might involve a new variablelilke PKG_TARGET_FULL which is set up properly in PluginSetup.cmake and only used in the metadata. But I havn't looked into the details.

leamas, 2020-11-18:
Closed (why not automatically?) by #154

PR #154 (2020-11-18): ci: Fix flatpak metadata <target> (#153).
With these changes the generated metadata should result in flatpak
and mingw plugins being visible and installable in the GUI. It
will also make the metadata comply with the plugin.xsd file file
at the plugins project.

Closes: #153

Please note that this PR is against the ci branch
commit 5e8d14e2 (2020-11-18): ci: Fix flatpak metadata <target> (#153).

With these changes the generated metadata should result in flatpak
and mingw plugins being visible and installable in the GUI. It
will also make the metadata comply with the plugin.xsd file file
at the plugins project.

Closes: #153
closed 2020-11-18 by leamas