opencpn-radar-pi/radar_pi #156: HALO24 fail with v5.1.4-ov51

electronsailor, 2020-11-29:
Unable to get A or B to properly initialize with radarpi v5.1.4-ov51 and two different up to date Windows 10 computers, though everything works swimmingly on a RPI4 running OpenCPN 5.2.4 and radarpi v5.1.4.

To begin with, the PPI for Halo A does not show up. The PPI for Halo B does show up. Selecting other Navico radar types is equally unsuccessful.

Thanks

[image]

keesverruijt, 2020-11-29:
Actually you can see it has found something on Halo A, the serial # is filled in. But it is still showing Scanning Interface.

If you use the context menu, can you ask it to show radar A control menu and then do "Show" there? It's probably just hidden.

electronsailor, 2020-11-29:
Yes, you are right. I can pull up the PPI for Halo A, but it still shows "No Radar" even though it has found the serial#. Simultaneously, the Zeus3 and the RPI both operate and view Halo A and Halo B normally.

keesverruijt, 2020-12-11:
We hope to fix this in an upcoming beta. I will update this once this is available.

keesverruijt, 2021-02-15:
We think this was caused by a hardwired check for serial numbers.

electronsailor, 2021-02-15:
Kees - Thank you. Douwe provided me with a preliminary radarpi that
bypassed serial number check and works with my Halo24.  Thanks for your
work on this!
Chip

alorman, 2021-03-09:
Piling onto this thread. Is it possible to post or email a release that bypasses the serial # check?
Thank you!

Hakansv, 2021-03-10:
@alorman 
We're working to release the beta version but the plugin upload process to OCPN plugin manager has a malfunction.
Meanwhile... If are you on Windows try this: [radar_pi v.5.2.0.beta4](https://www.dropbox.com/s/w8bgfnf56mmtez8/radar_pi-5.2.0-beta4-ov51-msvc-10.0.19042.exe?dl=1)
You know you have to uninstall a possible managed plugin in OCPN->Options->Plugins and restart O before a unmanaged plugin will be loaded?

keesverruijt, 2021-03-11:
v5.2.0-beta4 which fixes this is now available in the catalog. Enjoy! Please close this issue if this release fixes the problem for you.

alorman, 2021-03-11:
Can confirm! Thanks to everyone on this project for the awesome work

closed 2021-03-11 by keesverruijt