opencpn-radar-pi/radar_pi #157: Hardware installation (Raymarine)

xbgmsharp, 2021-02-15:
Hello

I have an old Raymarine C80 plotter connected with an old Radar radome.
From https://github.com/opencpn-radar-pi/radar_pi/wiki/Hardware-installation-(Raymarine)
`I understand a Seatalk hs port, which is nothing else than Ethernet.`

Should I cut the cable and have a RJ45 connect it to the RPI using the Ethernet port?
How should I wire the RJ45 color, I understand it is a crossover wiring?

Thanks

keesverruijt, 2021-02-15:
We don't know. We don't have direct access to a raymarine system. If you disconnect the plug, do you see a RJ45 jack inside?

keesverruijt, 2021-02-15:
Crossover or not went away as an issue with 100 MBit switches. Don't worry too much about that part.

xbgmsharp, 2021-02-15:
If I recall correctly, when unplug it does not look anything close to an RJ45.
Seem to be direct cable from the C80 the radome and power is provide by the plotter.

keesverruijt, 2021-02-15:
Good thing you did not cut the cable. It's not Ethernet, but an analog radome. 

I don't think the original C80 has a Seatalk_hs_ port, so you're out of luck. I will explain this on the Raymarine wiki page.

keesverruijt, 2021-02-15:
I did a little more searching and depending on what radome you have exactly and what you mean with C80 (Raymarine has at least 2 maybe 3 devices called this) it might be able to work, if the connector is Raynet.

Bloody Raymarine, so many connectors, devices (with the same name...)

sv-pizzazz, 2021-02-15:
Let me clarify: The only legacy chart plotter that works with Raymarine digital radars is the E80/E120 Classic that has an Ethernet port at the back. The port is a little odd in that you have to use a shielded metal connector to the switch (plastic connectors do not work) or an original waterproof Raymarine cable. If you have an E80/E120 connected to an analog radar, then OpenCPN with the Raymarine radar plugin can be connected to the E80 and will control the analog radar. It is nice that it works but at this stage it is so cumbersome you are better off using Raytech instead of OpenCPM. Digital radars (pre-Quantum) can be controlled directly. The C80 only has analog connectors and no Ethernet. The C90/E90 are later generation systems and they have only Ethernet ports, no analog radar ports. You can then control a digital radar directly. SV Pizzazz From: xbgmsharpSent: Monday, February 15, 2021 3:55 PMTo: opencpn-radar-pi/radar_piCc: SubscribedSubject: [opencpn-radar-pi/radar_pi] Hardware installation (Raymarine) (#157) HelloI have an old Raymarine C80 plotter connected with an old Radar radome.From https://github.com/opencpn-radar-pi/radar_pi/wiki/Hardware-installation-(Raymarine)I understand a Seatalk hs port, which is nothing else than Ethernet.Should I cut the cable and have a RJ45 connect it to the RPI using the Ethernet port?How should I wire the RJ45 color, I understand it is a crossover wiring?Thanks

xbgmsharp, 2021-02-15:
Thanks. By C80 i mean a C-Series Display (C80/C120). As I'm getting ride of the Raymarine plotter and replacing it by an RPI, I was open to get the radar data as well. I guess it is not possible. Thanks for the prompt support.

closed 2021-03-11 by keesverruijt