opencpn-radar-pi/radar_pi #160: After minimizing O, the radar goes into standby, after 30sec.

LeaseOnLife, 2021-03-11:
On Linux/Kubuntu, when I minimize the O window, either by clicking O in the taskbar, or use the minimize button, the radar goes into standby after 30sec, and then (obviously) loses all MARPA targets. When selecting another app from the taskbar and O is in the background (but not minimized), then the radar keeps running. I think the radar should be kept running after minimizing O.

This maybe related to, but seems different to:
https://github.com/opencpn-radar-pi/radar_pi/issues/141

O: 5.24, latest master, compiled
radar_pi, latest master, compiled

Hakansv, 2021-03-11:
@LeaseOnLife 
I've tested to minimize OCPN on my Windows and on one Rasbian, Stretch, and I don't see the same. So it would be something your Kubuntu system is (not) doing when minimized? 
On the other hand I never tried this before. A "silly" question: Why are you having a program Window minimized if you want use it and not just in the background when viewing another program?

LeaseOnLife, 2021-03-11:
@Hakansv 
there are several ways to get another program to the foreground. Minimizing O should be an acceptable option to a user. Add to that, widgets on the desktop a user may want to see.

LeaseOnLife, 2021-03-11:
I forgot to mention radar type: Navico 4G.
The 30sec seems to built into the radar: when radar is running and I close O, the radar stops turning after 30sec (I can hear that noise coming down the mast).

keesverruijt, 2021-03-11:
Hi @LeaseOnLife we are discussing the reason amongst the devs. We need to move our "keep alive" messaging to a different thread that is not influenced by the GUI window messages arriving (which, obviously, are not arriving for your minimized app).

keesverruijt, 2021-03-12:
This is the same issue as #141

closed 2021-03-12 by keesverruijt