opencpn-radar-pi/radar_pi #169: flatpak version : 5.2.1+994.v.5.2.1 installed but not working

filochard, 2021-05-18:
Hi 
I installed radar_pi
but the preferences button doesn't open the window to choose the radar
no icon appears in the icon tray

nohal, 2021-05-18:
Logfile?

filochard, 2021-05-18:
nothing interesting in the log file 
But : 
Even if the tarball is in the right place : 
/home/myusername/.var/app/org.opencpn.OpenCPN/config/opencpn/plugins/cache/tarballs/Radar-5.2.1-+994.v5.2.1_flatpak-18.08-x86_64.tar.gz
The lib doesn't appear in : 
/home/myusername/.var/app/org.opencpn.OpenCPN/lib/opencpn/
and its data subdirectory isn't created in : 
/home/myusername/.var/app/org.opencpn.OpenCPN/data/opencpn/plugins/

filochard, 2021-05-18:
Post Scriptum
Beside the flatpak opencpn that I am testing, I have a working opencpn (version 5.2.4) installed on my Linux OS (Mageia8) with plugins installed (I built rpms for them)

On this linux installed opencpn the radar plugin is OK (version 5.0.4-ov50) :  I can use the preferences button , the lib (libradar_pi.so) is in the right place, the icon is in the icon-tray, I can launch it and choose the radar to use...

leamas, 2021-06-16:
hm... excerpt from the log file:

```
11:08:06.450    INFO pluginmanager.cpp:2403 PlugInManager:  /home/mk/.var/app/org.opencpn.OpenCPN/lib/opencpn/libaisradar_pi.so
11:08:06.450    INFO pluginmanager.cpp:2404         Plugin common name: AIS Radar view
11:08:06.450    INFO pluginmanager.cpp:2406         API Version detected: 116
11:08:06.450    INFO pluginmanager.cpp:2407         PlugIn Version detected: 1.2
11:08:06.452 MESSAGE pluginmanager.cpp:1174 Checking plugin candidate: /home/mk/.var/app/org.opencpn.OpenCPN/lib/opencpn/libShipDriver_pi.so
11:08:06.452 MESSAGE pluginmanager.cpp:2124 Plugin is compatible by elf library scan: true
11:08:06.452 MESSAGE pluginmanager.cpp:2232 PlugInManager: Loading PlugIn: /home/mk/.var/app/org.opencpn.OpenCPN/lib/opencpn/libShipDriver_pi.so
11:08:06.453   ERROR dlunix.cpp:162 /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/mk/.var/app/org.opencpn.OpenCPN/lib/opencpn/libShipDriver_pi.so)
11:08:06.453 MESSAGE pluginmanager.cpp:2291    PlugInManager: Cannot load library: /home/mk/.var/app/org.opencpn.OpenCPN/lib/opencpn/libShipDriver_pi.so

```

seems like a strange linkage issue. It actually points to some problem in the build environment (?)

leamas, 2021-06-16:
Right. The 18.08 runtime has libstdc++-6.0.25, it doesn't seem to export GLIBCXX_3.4.26. It's a bit tricky, there is no "strings" command in the sandbox, just grep.

OTOH, the 20.08 runtime used in the beta branch (upcoming 5.6.0) has libstdc++.so.6.0.28 which exports also  GLIBCXX_3.4.26.  So this issue will be fixed when we update to the 20.08 runtime.  No point messing with current 18.08 since it is obviously not used.

PR #170 (2021-06-22): Flatpak: Fix 169, bugfix, add all anticipated 5.6.0 builds.
Current flatpak builds a does not load and are actually useless, see #169. We also have a need to build all flatpak versions planned for 5.6.0.  This PR adds a bugfix, some support for simplified local flatpak builds and a new build setup imported from shipdriver. 

New setup builds x86_64/18.08, x86_64/20.08 and arm64/20.08. At least both x86_64 builds load and seems to work. 

To test the 20.08 version one need the OpenCPN flatpak beta. Install and run using:

    $ flatpak install --user  https://flathub.org/beta-repo/appstream/org [... cut]
commit 34d06a8e (2021-06-22): ci: Update flatpak builds (#169).

Add a x86_64/runtime 20.08 build and refactor the build scripts.
These are changes imported from the shipdriver template. They
build all version anticipated for the upcoming 5.6.0 release.

Closes: #169
closed 2021-06-22 by bdbcat