opencpn-radar-pi/radar_pi #171: Radar overlay scaling problem.

ttlappalainen, 2021-06-21:
I am using latest OpenCPN and radar_pi with Navigo Halo. Is there any way to change scaling for overlay drawing? In picture you can see that objects does not fully match. In Nobeltec radar there was parameter to tune scaling.

[image]

douwefokkema, 2021-06-22:
I do not think that scaling is your problem here. But there could be an alignment error in the radar position and / or the heading. You should try the bearing alignment. Menu advanced -> Installation -> Bearing alignment. This will rotate your radar image, make it fit.

Douwe Fokkema

ttlappalainen, 2021-06-23:
No, it is not aligning problem. I have also tested that. Error is more clear on longer distances.

keesverruijt, 2021-06-23:
Sorry, we used to have a Scale factor coded in but that was dropped because we found that we did not need it any longer.I am in Horta and only have iffy charts for this area so I am not sure it is 100% correct but my picture seems to be OK. 

Do you have a chartplotter to compare the picture?
What charts are you using? Does the location match your boat, e.g. is the chart datum offset correct?

[image]
[image]

keesverruijt, 2021-06-23:
And can you check whether it gets worse or better if you change the _height_ of the OpenCPN chart window?

keesverruijt, 2021-06-23:
Which OS, what resolution, what size chart display?

ttlappalainen, 2021-06-23:
No other chart plotter. Charts are oeSENC Finnish vector charts - should be very good. Vessel position matches well. GPS compass gives <2 m longtime accuracy. I can not say specially on this point position matching with that accurate. I took this just sample picture at anchorage, but there is same effect during motoring or sailing.

W10, 1920x1080. Display is 15" - or what you mean.

I can not say from your pictures, do you have same effect or not. It is easier to see, if you have refrective floting boya in low water so that they do not move more than few meters or fixed boya on open sea. If you e.g. forget boya on my picture, it looks OK.

douwefokkema, 2021-06-23:
Which distance is right, distance on chart or distance on radar? I suggest you measure distance to a clear target on the chart. Also measure the distance to the radar return of that target using the VRM in the radar window. The same? Then the mapping of the radar image on the chart is wrong. Different? Then find out which one is right, use google satellite maps or other. From that we should be able to proceed. If the radar is wrong we might need an adjustable scaling to correct.

Douwe

ttlappalainen, 2021-07-01:
I can not say which distance is right and it is a bit difficult to measure. If I take any system, which draws something to screen and try to measure there, how do I know that it has right exact scaling. Also what ever draw scaling is on OpenCPN I'll expect it uses same scaling to calculate position back and so will show right distance to object according to chart. Only sure thing is that I see a small difference in chart and radar scaling.

Below is an other picture, which shows that all objects around me are slightly more near on radar image than on chart. This is still not perfect, since boyas on N and S are still floating ones. I have not passed fixed boyas lately to get sample from that, but have done before and seen same effect.

[image]

keesverruijt, 2021-07-01:
It could be different per individual radar.

An overall scaling factor, even if you have to set it in the config file, might not be such a bad idea. At least Timo could experiment to see what we are talking about (2%?)

ttlappalainen, 2021-07-01:
1-3%, what I have measured. When the is scaling it would be easier to test and follow, is there differencies between ranges. I have not went deep in to the radar systems and so do not know what can cause this.

ttlappalainen, 2021-07-01:
Just discussed with guy who makes Plotter/Radar installations. He told that there is setting on plotters to tune overlay scaling called with different names - was someting like "Display timing". After installation he normally does tuning on specific place, where he can see on 1.5 nm distance well known mark.

keesverruijt, 2021-07-01:
My Navico plotters do NOT have this functionality.

Hakansv, 2021-07-01:
Timo. I was earlier commented this timing function to likely be the one used during installation to adjust for the length of the antenna cable on e.g. a x-ray radar. So not applicable here. 
But if it's like that different Halo units may differ in there range report an adjustment factor would probably help. 
For a long boat where the GNSS receiver is far from the radar scanner it could also be useful for the chart overlay function although the adjustment calculation will be more tricky?

LeaseOnLife, 2021-07-01:
I can confirm that on my old 4G I also have all objects 1.5-2% too close, all around.

Hakansv, 2021-07-01:
I haven't measured that rigorously on my 3G but one question to you;
When you claim a false range from the scanner. Is it for the chart overlay only or also for the PPI window?
Another Q: does the radar extreme range ring show a correct range?

douwefokkema, 2021-07-01:
I could make a scaling control under installation. Going +/- in steps of 0.1% (using +/- 1 and 10), to max 10%. This could work on the pixels per meter variable, and function both on overlay and radarwindow. Saved and restored with the ini file. Could be implemented for all Navico radars and easy to add to other radars. Is that what we would need? Next weeks I will be on the boat, could test on my 4G.

Douwe

LeaseOnLife, 2021-07-01:
Hakan, not sure how to measure that well enough in the PPI window... 

regarding range rings: on my system, the range ring was always about 1.5x the range I set as radar-range. radar returns are shown beyond the set radar range, up the that range ring. Example, radar range set to 6nm, the ring is at ~9nm radius.

Looking closer though, there is a 2-3% error there too. at 6nm range, my ring shows at 8.7nm. at 24nm radar range the ring shows at 39.3nm. could be a coincidence or related ?!?

Hakansv, 2021-07-01:
To measure in the PPI you can put a mark on a target and read the distance. Compare that to a the real distance from a chart.
I've to check my extreme range ring when back to the boat next time to check the same.

ttlappalainen, 2021-07-02:
+-1% scaling control would be perfect. I am very interested to test on my Halo.

douwefokkema, 2021-07-02:
ttlappalainen,
An experimental version with a "Scaling" button under Advanced->Installation is now in branch scaling (https://github.com/opencpn-radar-pi/radar_pi/tree/scaling). Scaling is done in steps of 0.1% with the +/- 1 and 10 button. Do you compile yourself or do you need a compiled version?

Douwe

ttlappalainen, 2021-07-02:
I need compiled version for Windows 10 please. I have not setup OpenCPN or plugins environment. Unfortunately too much other programming work to do.

douwefokkema, 2021-07-02:
Setup package at: https://www.dropbox.com/s/lfmfi8cv8u45xv8/Radar-5.2.1-%2B2107021015.978b1bd_msvc-10.0.19043-win32.exe?dl=0
First uninstall the managed version if you have it installed.

Douwe

LeaseOnLife, 2021-07-02:
The scaling works well here on my old 4G, tested only with short range until we leave this anchorage. A scaling of +25 (2.5%) seems best. It also 'corrects' the extreme range ring: when at a radar range of 1/2 mile, the range ring is at 0.75 miles (not 0.73 as it was before).

ttlappalainen, 2021-07-02:
Perfect, thanks!. Now it looks with 3% setting how it should be. On second picture upe-left there is two sma ll islands and fixed mark. From fixed to NE mark there is floating right buoy.

[image]

[image]

douwefokkema, 2021-07-02:
Thanks to the architecture of radar_pi these kind of jobs are quick and easy, all honors to Kees Verruijt for the design of this system. I could merge this to master as is, waiting for the next release. But is the naming right, should there be any cosmetic changes? Stepping in integer units of 0.1%  is may be not so nice, but seems acceptable for the installation menu. Should this work for all radars? It doesn't do any harm if you don't touch it.
Douwe

ttlappalainen, 2021-07-02:
In GUI it would be naturally better to show scale change in % e.g. 2.9% instead of plain value 29. 0.1% as step is good. It is difficult to see is 2.5% better than 3.0 %. So I see that 0.1% step is fine. 0.5% would probably be too rough for someone.

Also we need new strings to Crowdin. Scale in English is probably not the best. Should be something like "Scale correction".

keesverruijt, 2021-07-02:
Douwe, thanks man for picking this up. I think this might be useful for all radar types/brands?
Timeo, I agree that a different presentation of the scale correction (e.g. "2.9%" instead of 29) is nicer, but due to the generic nature of our controls this is a whole heap more work. I suggest we leave it at this for now.

keesverruijt, 2021-07-02:
> regarding range rings: on my system, the range ring was always about 1.5x the range I set as radar-range. radar returns are shown beyond the set radar range, up the that range ring. Example, radar range set to 6nm, the ring is at ~9nm radius.

Yeah. This is a weird artifact of the Navico radars, I think they do this so you can have an "offset" center without changing the radar range.

> Looking closer though, there is a 2-3% error there too. at 6nm range, my ring shows at 8.7nm. at 24nm radar range the ring shows at 39.3nm. could be a coincidence or related ?!?

If we all have this then the default correction for Navico radars might need to be != 0 or we change the calculation internally? I will try tomorrow with this branch, but I don't have any nice small targets with well-defined locations here, and I can't go out 'cause I will lose my spot in the marina.

Hakansv, 2021-07-03:
Tested on my 4G and now when looking closer there was a small range error also on my system. A +2.5% (25) of the range scaling does a fair correction. 

The extreme range ring has the same error as others.

UI: 
I can live with the ten times value in the control. I think most users would just adjust to a value making a correct range. No matter of the figures.
Maybe we should name it "Range scaling" or something to make it clear it's connected to the range?

For me we could merge to master.
When done I can administrate Crowdin issues.

Thanks to Timo for issuing this and to Douwe for a quick development.
Håkan

Hakansv, 2021-07-03:
BTW: I suppose this scaling would be useful on all radar system if possible

douwefokkema, 2021-07-03:
I did some updates. Instead of "Scaling" "Range adjustment" is used. Value displayed is now like 3.5 %. For this I made an exception in the software, not nice from a programmers poit of view, but the display looks fine. New version at the dropbox above.

ttlappalainen, 2021-07-03:
Thanks, but still shows "Scaling", number 30 and link is exactly same. Did you provide old link?

Hakansv, 2021-07-03:
Douwe..
Looks fine. And also tested on my BR24 on the RPi4.
For those who prefer to Import the test by the plugin manager here are two links.  :)
(Win and RPi Buster)
https://www.dropbox.com/s/s8w8xg73ao7a82c/Radar-5.2.2-Beta%2B2107031222.f6e442d_msvc-10.0.19042-win32.tar.gz?dl=1

https://www.dropbox.com/s/2bh80o0b1ck86pa/Radar-5.2.2-Beta_raspbian-10-armhf.tar.gz?dl=1

douwefokkema, 2021-07-03:
> 
> 
> Thanks, but still shows "Scaling", number 30 and link is exactly same. Did you provide old link?

I checked my link on Dropbox, it is the new version, in the log: radar_pi.cpp:256 5.2.1-+2107031056.f6e442d 
where the f6e... is the last commit. And it shows 3.5 %.

douwefokkema, 2021-07-03:
And I added the Range adjustment for all radars. Won't harm when not used. Not yet in the links above but in branch scaling.

Hakansv, 2021-07-03:
Douwe..
Still fine on Win and RPi. 
So now also for the Emulator, FWIW :-)

ttlappalainen, 2021-07-03:
Thanks. Now it works. I installed it with manager from Håkans link. Great work from all!

Is "Range adjustment" right term? In a way you set Range how far you measure. From "Range adjustment" I get impression that I would tune it to see 1.8 nm instead of 2 nm. Isn't this some kind of correction term for some unknown(?) error? So I would rather use "Range correction".

douwefokkema, 2021-07-03:
> Thanks. Now it works. I installed it with manager from Håkans link. Great work from all!
>
> Is "Range adjustment" right term? In a way you set Range how far you measure. From "Range adjustment" I get [...]

Ye, I thought about Range correction as well. But Range correction implies that the range shown is wrong, actually I don't know, It could be that the projection of the chart on the screen is wrong. The only thing I am really sure of is that the range shown is adjusted.

douwefokkema, 2021-07-05:
The Range adjustment modifications have been merged in master, branch scaling is removed.

Hakansv, 2021-07-05:
Very good. I'll during the day upload a new pot-file to Crowdin.
BTW: The CI fail I think depends on the same version is used again. I'll change version when I merge the new pot file for a check.

douwefokkema, 2021-07-05:
OK, then I will close this issue.

closed 2021-07-05 by douwefokkema

Hakansv, 2021-07-05:
Now also is the fresh pot file uploaded to Crowdin.