opencpn-radar-pi/radar_pi #172: Question: Halo3 doppler speed and velocity track

kizilkayabayram, 2021-06-25:
According to issue #133, As far I as understood halo 3 radar does not transmit any doppler measurements. Instead it returns only approaching/receding data. How this relates the velocitytrack feature? Does velocitytrack provides separate doppler speed for each point or only returns approaching/receding info? 

I'm actually looking for a way to get velocity values for each measurement from radar data and saw velcoitytrack on/of pcap file in examples folder. 

Thanks in advance.

Hakansv, 2021-06-25:
@kizilkayabayram 
You can use the (M)ARPA function in radar_pi. 
Either set up an guard zone from the area you want to collect ARPA targets and select "ARPA On" for that zone.
Or right click near desired target and select "Acquire radar target" to start the MARPA function for the nearest target.

A (M)ARPA target acts like any other AIS target but with a round symbol. See picture
[image]

kizilkayabayram, 2021-06-30:
Are those measurements are based on doppler measurmenets?

Hakansv, 2021-06-30:
All is calculated by radar_pi and has nothing to do with the Doppler function. So the ARPA function can be used with any radar scanner supported by radar_pi.

keesverruijt, 2021-06-30:
> According to issue #133, As far I as understood halo 3 radar does not transmit any doppler measurements. 
> Instead it returns only approaching/receding data. 
> How this relates the velocitytrack feature?
This _is_ the VelocityTrack feature. 
> Does velocitytrack provides separate doppler speed for each point or only returns approaching/receding info?
Only approaching/receding info, which is a single value for every 'radar pixel'. 

> I'm actually looking for a way to get velocity values for each measurement from radar data...
This is not possible.

kizilkayabayram, 2021-07-01:
Thanks a lot for quick and informative responses.

closed 2021-07-01 by kizilkayabayram