opencpn-radar-pi/radar_pi #174: I may have some access to a Raymarine Quantum Chirp radar.

trudK45, 2021-06-29:
During the autumn and winter I might have access to the named radar and may have the time to do some work figuring out the commands/returns. This particular radar is set up with RayNet cable connection.
Any suggestions of how to interface a PC with Wireshark ? Can you just connect the PC with WiFi to the radome as a second  interface? Anyone here know the answer to this ?

keesverruijt, 2021-06-30:
Wire everything to a standard RJ45 switch?

douwefokkema, 2021-06-30:
I have a working version of radar_pi supporting Raymarine Quantum radar. It has been tested partly, connected to a MFD that was wireless connected to the radar. Direct WiFi connection should work as well. This version is in branch quantum. Further testing is required. Recording commands is kind of tricky, the Raymarine switch transmits the UDP unicast commands only to the radar, so it is hard to get them.
Douwe Fokkema

trudK45, 2021-06-30:
@douwefokkema
As mentioned earlier the radar in mind is connected through cable. Do you know if the commands are echoed back on the WiFi connection ? If not shouldn't it be possible to insert a mirroring ethernet switch between the MFD and the radar radome and thus see all packets to/from the radar on a PC ?

douwefokkema, 2021-06-30:
If the radar is WiFi connected, commands are sent over the WiFi to the radar. But using Wireshark we have not been able to get these commands. If radar is wired to the MFD, you can put a switch in between, but a regular switch will transmit only to the address indicated and not to you Wireshark machine. You will need a hub or a programmable switch and know how to program it. That is what Al Grant in New Zealand did until his switch broke down. We now have all commands that I know is there (may be there are more), implemented and functioning. Only gain is not working, no solution yet, a fresh recording of the gain command would be useful. You are welcome to participate in the testing, if you compile yourself, you can get the software. Otherwise I can send you an installation package. You could mail me directly at df@percussion.nl
Douwe

keesverruijt, 2022-04-08:
@douwefokkema was able to add Quantum support using other means.

closed 2022-04-08 by keesverruijt