opencpn-radar-pi/radar_pi #178: What support is there for saving radar data for post-processing, replay?

ForceConstant, 2021-09-29:
I was wondering if with using this plugin, can you save the radar data to look at it again? Either in or out of opencpn? 

Thanks.

Hakansv, 2021-09-29:
@ForceConstant 
No, there's no such plugin function. 
For occasional use the Wireshark program to record and on Linux, (Mac?) Tcpreply to replay the saved file is what's used for development here.

douwefokkema, 2021-09-29:
And if you also output your position and heading info (while sailing) to UDP / Ethernet, you will record (with Wireshark) all information about your trip combined with the radar data. You can replay your voyage completely by replaying the .pcapng file from Wireshark with TCPreplay. But be prepared for many gigabytes of data.

ForceConstant, 2021-10-04:
ok thanks.

closed 2021-10-04 by ForceConstant