opencpn-radar-pi/radar_pi #180: Probably an opencpn thing, but how to install/enable plugin without internet?

ForceConstant, 2021-10-04:
I am trying to install/enable this radar plugin, but it tries to update the master catalog which fails without internet. Is there something I can do to fix this short of connecting internet?

Thanks.

Hakansv, 2021-10-04:
@ForceConstant 
You can catch the "Tarball" and save it to a stick and use the Options->Plugins - "Import Plugin" to run it on the offline machine
.
One way to find the Tarball is to on a system with internet "Update catalog master" and then find the radar_pi Tarball at the same folder where the opencpn.log resides. Go to subfolder:  /plugins/cache/tarballs.
This provides the online machine use the same OS as the offline target machine.
See more here: https://www.cruisersforum.com/forums/f134/opencpn-v5-1-managed-plugins-234223.html

OR..
What's your OS on the target machine?
I could possibly attache a tarball here.

ForceConstant, 2021-10-04:
@Hakansv  thanks for the reply, I am actually building radar_pi myself, so have the tarball, but when I select install I get "Can't access current catalog" unless i have internet.

Hakansv, 2021-10-04:
re: " when I select install " I may misunderstand but what's "Install"?
You have to use the pluginmanager "Import plugin" method.
[image]

ForceConstant, 2021-10-04:
@Hakansv yes sorry, I meant "Import Plugin..."

Hakansv, 2021-10-04:
Hmm.. Strange.
Here I've my .gz Tarball on a USB-stick. Don't you get the same view?

[image]

Hakansv, 2021-10-04:
Never seen that.
What's your system?

ForceConstant, 2021-10-04:
Ubuntu 20.04

Hakansv, 2021-10-04:
OK, I don't have any such. Try to download this to the stick and check for a difference:
https://dl.cloudsmith.io/public/opencpn-radar-pi/opencpn-radar-pi-prod/raw/names/Radar-5.2-ubuntu-gtk3-20.04-tarball/versions/5.2.2-+1316.be299ce/Radar-5.2.2-+1316.be299ce_ubuntu-gtk3-20.04-x86_64.tar.gz

bdbcat, 2021-10-04:
I cannot find this string in the 5.5.x source code:
""Can't access current catalog""

So, we are clearly missing something obvious.
Are you using OCPN 5.5.928?

Dave

ForceConstant, 2021-10-04:
Actually it is 5.2.4, which is the default as per the instructions from ppa:opencpn/opencpn Yeah that is a year old.

bdbcat, 2021-10-05:
OK, it seems that you have no "ocpn-plugins.xml" in your OCPN data directory.  
Strange, but anyway....
Create an empty file that looks like this:

```
<?xml version="1.0" ?>
<plugins>
  <version>0.0.0</version>
  <date>2021-09-29 23:56</date>
</plugins>

```

Put that file in the same place that your OCPN logfile lives.
Good Luck
Dave

closed 2021-10-19 by ForceConstant