opencpn-radar-pi/radar_pi #187: Android builds blocked (GLES vs OpenGL).

leamas, 2021-11-21:
Buildińg on Android  requires using  GLES rather than the full OpenGL interface. As I understand the situation, radar_pi uses functions not available in the GLES subset and is thus not possible to build on Android. 

It seems that at least parts of this can be resolved by using external libraries, basically GLES add-ons for missing OpenGL functions. However, these things are basically above my paygrade in this project.

Hakansv, 2021-11-21:
Basically we need a Ethernet wire to connect "our" supported radar scanners to OCPN so no need to build for Androids. I agree. 
I use one in my boat also for the radar view but then it's connected by VNC. :)

sv-pizzazz, 2021-11-22:
Dear Developers, First, I highly appreciate everything done on the radar-pi plugin and I am happy for what you have developed so far, so only respect for you.  In terms of the specific experience with Desktop Remote or VNC to a computer running OpenCPN, there is a lot of room for improvement. I am running Remote Desktop to an Intel NUC to a Raymarine digital radar dome. Yes, it works but the steps are very geeky… start the NUC, launch OpenCPN, fiddle with the mouse to set everything up on the NUC, then go to the flybridge, start the tablet, the screen resizes, the radar window disappears then re-appears. The Intel NUC connects wirelessly to the router than bridges the radar dome with the Raymarine C120W, so I have no issues with wired vs. wireless so far. It seems that the preferred usage for OpenCPN is below, in a protected space, with a mouse where one has access to the full functionality, wind, route planning, navigation, etc. However, with the ease of use of apps like Navionics, it seems that very few people do this. People just check the weather on Windy, launch Navionics and off you go. You do route planning on the way. If we can have radar on the tablet/phone direct, a lot more people will just use OpenCPN Android instead of a second chart plotter for the flybridge. That was my point. On a separate note, there is an app called Raytech Navigator that can control Raymarine digital radars (or sonars) from the PC. I often use both Raytech and OpenCPN side by side. Raytech is a lot harder and non intuitive to use but one of the benefits to you guys may be that you can issue all sorts of commands to the radar (setup, curves, MARPA stuff, etc.) and it should be a lot easier to capture the output of the app on the PC than to have to use a chart plotter connected to the radar with a sniffer. If you believe the program may be useful to your development effort, I will be happy to share my pass code with one of you to help you with debugging and development. The program can be downloaded from the Raymarine website. You will not even need a radar dome to do the coding, just copy the output of the program. Thank you for everything, just trying to be helpful. Vassil ---Vassil Shtonov+359 88 6059129 From: Hakan SvenssonSent: Sunday, November 21, 2021 11:39 PMTo: opencpn-radar-pi/radar_piCc: SubscribedSubject: Re: [opencpn-radar-pi/radar_pi] Android builds blocked (GLES vs OpenGL). (Issue #187) Basically we need a Ethernet wire to connect "our" supported radar scanners to OCPN so no need to build for Androids. I agree.I use one in my boat also for the radar view but then it's connected by VNC. :)

keesverruijt, 2021-11-22:
Agree with @Hakansv there is very little to no use if we would bring out the plugin on Android. We will only disappoint 90% of the users that have radars that work via multi-cast, which doesn't lend itself to WiFi. Only the (lousy) WiFi radars would work. Since none of the devs has such a radar, developing this would be ~~a pain in the~~ very difficult and we are of the opinion that you should not be using O this way anyway.

closed 2021-11-22 by keesverruijt