opencpn-radar-pi/radar_pi #190: Halo24 data UDP rates

pepperlk, 2021-12-15:
According to SIMRAD the Halo24 can do "60 RPM operation at ranges up to 2nm" leading to +-1 sec per set of data packets.
Connected directly to the radar no switch in between or anything both WireShark and radar_pi are giving me 15ms per packet of 17160.
If I do the math 15 * 128 = 1920ms and this plugin and all other UDP measurements agree.

Is this the udp protocol speed being seen by other units or is it just mine?

keesverruijt, 2022-04-08:
You can switch the _Advanced > Scan Speed_ menu item to High or Medium depending on range. The plugin doesn't really know which speed is possible at which range, just try changing it; it tries and if it fails it falls back to a value allowed by the radar.

keesverruijt, 2022-05-12:
I did this on my own Halo24 and indeed it seems it doesn't seem to hit 60 RPM.

pepperlk, 2022-05-12:
The rotational speed when i took the dome apart and check is 60 ish rpm and im sure you can hear it... 
BUt the only way i have been able to get 60 rpm of data from it has been to activate radar a and then 500 ms  >< 900ms activate b.. then i get rotations of data coming in at 1.1 to 1.2 sec

But both radar units have to be processing at the same time to achieve this.

commit 3b0a3dd2 (2022-06-12): Fix #190: Show RPM on info window
closed 2022-06-12 by keesverruijt