opencpn-radar-pi/radar_pi #192: Can't install

eov-oleg, 2022-03-17:
Hello!
I try to install opencpn-radar-pi to RaspberryPI  (Raspberry OS buster.)
How to fix this error?

-- *** Staging to build rmradar_pi ***
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindwxWidgets.cmake:1008 (find_package_handle_standard_args)
  cmake/PluginConfigure.cmake:60 (FIND_PACKAGE)
  CMakeLists.txt:22 (INCLUDE)
-- Configuring incomplete, errors occurred!

Hakansv, 2022-03-17:
If you have OCPN 5.6 running use the Plugin manager to install radar.
But.

And if you're interested in raymarine try the beta catalog instead of master. (Tweak opencpn.conf as advised.)
If you for other reasons want to build yourself se the [install ](https://github.com/opencpn-radar-pi/radar_pi/blob/master/INSTALL.md) file here.

Hakansv, 2022-03-17:
And BTW... rrmradar is not this project

closed 2022-04-08 by keesverruijt