opencpn-radar-pi/radar_pi #203: HALO 20 reduced features not reflected in UI

keesverruijt, 2022-06-14:
The HALO20 works but we show all features that it doesn't have (2nd range, Doppler, maybe more.)

keesverruijt, 2022-07-04:
Can some HALO20 users please do the following:

1. Update to the latest Alpha release, e.g. the most recent tarball at https://cloudsmith.io/~opencpn-radar-pi/repos/opencpn-radar-pi-unstable/packages/
2. Stop OpenCPN, edit the ini file and set VerboseLog to 73.
3. Start OpenCPN, and test all buttons (or at least as many as you can) on the Adjust, Advanced and Installation pages. Make a note when you first see a setting change when you click the button, and then it immediately changing to something else. This means that the radar doesn't support that setting: the plugin sends the command for that value, but it doesn't work and the radar reports the old setting.
4. Stop OpenCPN.
5. Report your findings here or in an email to me.
6. Upload the opencpn.log created with the VerboseLog at 73.
7. Now you can revert the VerboseLog to 0.

svScoots, 2022-07-06:
Hello,
 I ran the HALO 24 through the menu items, the only issue I found was when the sea-clutter was in manual mode, it started at a value of 40.  When I hit +10 it went to 50 but then went back down to 41 after  few seconds.  It is repeatable.  I did this during the run through the menu options and repeated it at the end.
[opencpn.zip](https://github.com/opencpn-radar-pi/radar_pi/files/9053027/opencpn.zip)

keesverruijt, 2022-07-21:
Ok thanks @svScoots now waiting for a HALO20 user to do the same actions and upload the results.