opencpn-radar-pi/radar_pi #205: ARPA is not initiated for first time users.

Hakansv, 2022-06-19:
I've observed this for some radar_pi first time users.:
If radar plugin is installed for the first time on a system neither (menu acquire..) MARPA nor (guard zone) ARPA is initiated.
The contour around a target is not drawn even if e.g. a guard zone clearly would start to process an isolated target. This is easily tested by use of the Emulator "zoomed out" rectangles. 
But.. if I close and restart OCPN an ARPA target is directly identified when needed.
I get the impression some config setting is missed for the ARPA logic at the first time use of the plugin and we need to save to config more often? For example is a setting like "EnableCOGHeading" not saved to config until plugin DeInit.
I'll try to find out but you may directly know what's missing?

Hakansv, 2022-06-21:
This is strange..
The first time use from a opencpn config cleaned from all [PlugIns/Radar] settings.
ARPA is not detected most probably due to a unreasonably value for "Radar0MinContourLength".
When O is closed after installation the "Radar0MinContourLength" can have different config settings making a ARPA recognition impossible. I've seen Radar0MinContourLength=303, and Radar0MinContourLength=1647262819.

After a O restart the Radar0MinContourLength is changed to "6" as stated in:
 ```
pConf->Read(wxString::Format(wxT("Radar%dMinContourLength"), r), &ri->m_min_contour_length, 6);
if (ri->m_min_contour_length > 10) ri->m_min_contour_length = 6;  // Prevent user and system error
```
Then is also ARPA detection working.

We may need to run a proper initialization after plugin installation?

douwefokkema, 2022-06-22:
Yes, this is indeed strange. It seems that in case the opencpn.ini is not present, none of the pConf->Read( statements in radar_pi are executed. So m_min_contour_length remains uninitialized. So better initialize m_min_contour_length = 5; in radar_info. I will add this.

Hakansv, 2022-06-22:
I tried this:
```
[@@ -264,8 +264,10 @@ int radar_pi::Init(void) {
     m_radar[r] = new RadarInfo(this, r);
     m_settings.show_radar[r] = true;
     m_settings.dock_radar[r] = false;
     m_settings.window_pos[r] = wxPoint(30 + 540 * r, 120);
+    RadarInfo *rinit = m_radar[r];
+    rinit->m_min_contour_length = 6;
   }
 
   m_GPS_filter = new GPSKalmanFilter();](url)
```
And that seems to do the trick. I'll do some more tests with a real "clean" system.
If nothing better comes up I can merge..

Hakansv, 2022-06-22:
I tried a tarball to a cleaned system and now ARPA detection is working as expected.
I can wait a day or two for possible better ideas how to code. Otherwise I'll merge then.
@douwefokkema should revert your change is I merge?

douwefokkema, 2022-06-22:
About my changes: these are initializations that won't do any harm and may be useful in some cases. You could consider to leave them as they are.

Hakansv, 2022-06-22:
OK, will do. Done some more testings and it seems to work.
I'll merge it as version 5.4.1 alpha
Thanks

commit 98803b4c (2022-06-23): Fix #205: Fixup of previous ci

1. Resetting value to 6 doesn't help; prove this with assert.
2. Update Changelog.md.