opencpn-radar-pi/radar_pi #209: Trails Off is not saved correctly.

douwefokkema, 2022-07-10:
After start up trails is always at 15 seconds. Trails can be set off, until shutdown and start up, then On again. 
Recorded with 4G radar on Windows 11. Can be reproduced without radar connected.

Hakansv, 2022-07-10:
The same is also valid for Radar Overlay Canvas
If e.g. Config file "Radar0OverlayCanvas1"=1  the resulting status after a restart is Off (-1).
If I change the settings to "On" the write to config method write the correct value in the file.
So, read and write from and to the config file is probably correct but update settings goes wrong somewhere.

keesverruijt, 2022-07-10:
Probably a side effect of the change I made to have the 'real' radar controls work on the first click.

keesverruijt, 2022-07-10:
That is issue #195.

douwefokkema, 2022-07-12:
The following update in ControlsDialog.h seems to solve this issue:
@@ -610,9 +610,7 @@ public:
 
         this->SetFont(m_parent->m_pi->m_font);
         m_item = item;
-        if (m_ci.names) {
-            SetState(RCS_MANUAL);
-        }
+        
         UpdateLabel(true);
     }

I have not pushed the update as these statements may have an other purpose.

douwefokkema, 2022-07-12:
The problem is that the m_control_dialog->Create( is done after the reading of the ini file, these statements override the values in the ini file.

keesverruijt, 2022-07-21:
Ah, I was confused by "solve the issue", but those are changes I made that _cause_ the issue. 

It should work again now.

closed 2022-07-21 by keesverruijt

Hakansv, 2022-07-21:
When building on Windows using cmake --build . --target tarball I get the below message after install. There's normally no file permissions involved on Windows. Strange!
I'm in the boat on my way home so no further details for the moment.

[image]

Hakansv, 2022-07-21:
One thought:
I'm using Windows and OCPN code from recent GitHUb master.
@leamas Could anything in Pluginmanager be the cause?
I'll try to debug when home.

keesverruijt, 2022-07-21:
I don’t know about windows but on Mac and Linux the build step creates two libraries, one in the build directory that you cannot use, and one deep down in the hierarchy under the build directory. That is the one to use. 

Met vriendelijke groeten, kind regards,

Kees Verruijt

Hakansv, 2022-07-21:
I created the tarball as usual. Then imported that to OCPN using the Pluginmanager. For radar_pi the above message appeared.
To double check I did the same for o-charts plugin and the same method yielded the same error. Obviously some bug in the latest code to OCPN. I'll try to bisect.

Hakansv, 2022-07-21:
Sorry my bad..
I forgot the default build is Debug. When build using also "--config RelWithDebInfo" importing the tarball works as expected.

And the corrections for this issue seems to work fine. The trails settings in the config file are now reflected after a O restart.
I've to check the rest when back in the boat.
Thanks
Håkan

Hakansv, 2022-07-22:
Now tested on the 3G. Trail settings and sea clutter preferences are now loaded from config file at start up as well as reflected in the menu after a run time change. Very good!

Another issue maybe connected to this:
Trail color changes are not reflected at run time. 
If I change start or end colors nothing happens. If I stop and restart O the changed colors are then used, so it's read from config but not changed "on the fly".
Håkan

Hakansv, 2022-07-22:
Correction about trail color changes.
If I change color and then switch trails off and then on again the color is changed.
So maybe understandable and not much to spend efforts on?

commit 8a8e7a99 (2022-07-21): Fix #209: Repair button behavior broken in #206