opencpn-radar-pi/radar_pi #212: Garmin Fantom 18

dpeckham, 2022-10-12:
Hi, I just bought a brand-new Garmin Fantom 18, which might have been an expensive mistake. I am having no luck getting it to work on my Rasberry Pi 4 with OpenCPN 5.6.2. It just sits at "scanning interface" and I never get any status lights on the radome. I have tried shorting pin 5 to ground, but it doesn't ever seem to do anything. Anyone have an ideas?

Yes, I know this isn't a software bug/issue, but the community is pretty small. I am a developer and have this running in my home lab at the moment, so i can help with any troubleshooting or other information gathering if needed. 

Thank you!

dpeckham, 2022-10-12:
Did you use this one? Garmin 010-00959-00 GMR 18 xHD

I don't mind putting in the work for the Fantom, but I don't want to buy a Garmin chartplotter. I do have an oscilloscope and all the other gear needed to reverse engineer. But not the $1500 for a plotter :-)

Dave

keesverruijt, 2022-12-11:
@dpeckham any luck yet, do you need any help with the source code?

dpeckham, 2022-12-12:
> @dpeckham any luck yet, do you need any help with the source code?

Hi. I have all the hardware but won't be able to work on it again until February as I will be crossing the Atlantic in January. I was able to get it partially working already, so I am sure it's going to work with a little effort.

fluffy, 2023-01-09:
Looking forward to follow the progress on this.