opencpn-radar-pi/radar_pi #214: Kalman Filter issue

oscar-polheim, 2022-10-31:
I've compared Kalman Filter used in radar_pi with the [one](https://github.com/lacker/ikalman) I use in my projects with GPS.

From what I can see, radar_pi's KF seems not so good even compared to raw GPS.

Legend:
[image]

Sample is from a boat GPS at 10Hz:
[image]

Is this something known or expected? Or should I dig deeper and try to understand if I did something wrong using radar_pi's KF.

douwefokkema, 2022-11-01:
Very interesting. But please clarify the image for me. I understand the blue one is the original raw GPS. Is the pink one your KF? The green one is marked OpenCPN, but is this the track from OpenCPN or the results of the KF in radar_pi?
The KF in radar_pi is only used to get the vessel position on a given time (could be in the future), not used otherwise by OpenCPN.
Douwe, alternatively mail me at df@percussion.nl

oscar-polheim, 2022-11-02:
The pink one is the result of KF from https://github.com/lacker/ikalman
The blue one is the result of KF from radar_pi

douwefokkema, 2022-11-02:
It is already several years ago that I made this. The filter used in radar_pi was optimized for slow GPS (1 Hz) and typical sailboat speeds, with a slow turn rate (some 20 degrees/sec). With a rapid update GPS a Kalman filter is not needed at all, the last GPS position is sufficiently accurate for "now".

oscar-polheim, 2022-11-03:
Looking at some preliminary data with 1Hz, it looks better. What would you do to adapt the filter to 10Hz?

douwefokkema, 2022-11-03:
I would remove it. A position not older than 0.1 sec is sufficiently accurate for our purpose.
What is the scale of your images? What speeds are used? What turn rates?