opencpn-radar-pi/radar_pi #218: Problem with ARPA target on radar Halo 20+

rdouguet, 2022-12-15:
Hello,

I 'm using OpenCPN v5.6.2 (on windows) and the radar plugin v5.3.4 with an Radar Halo 20+.

I set the computer ip address to communicate with the radar and I 'm able to start/stop the radar and to see correctly the data on radar view (Navico Halo A).  Now, I want publish the detected target (ARPA) on UDP link and there is I have a problem :

When I enable the ARPA mode for a Guard zone 1 or Guard zone 2, no ARPA data are detected (normally ARPA targets are circled in bold in radar view and the targets appear on seamap) => today, I have nothing and if I check on NMEA consol, I dont see the RATTM frame corresponding to ARPA targets)/

Has anyone had this problem before ?

I already used this radar with OpenCPN a few months ago and everything was working fine... So I don't know if this problem caused by the last version of OpenCPN or last version of radar plugin..

Thanks for your help,
Ronan.

Hakansv, 2022-12-15:
Ronan..
I see it in NMEA Debug window as of below.
One Q: Do you have a valid position? It's needed for ARPA functionality.
Also in resent master HDT is needed but that's not yet published so wouldn't be the trouble?
Once in the OCPN multiplexer it's no problem to send it out as requested.
```
<GREEN>15:35:17 (virtual) $RATTM, 1,1.033854,237.500000,,5.76,176.3,T, , ,N,ARPA 1,T, *22<0x0D><0x0A>
<CORAL>--> 15:35:17 (TCP:192.168.1.168:10120) $RATTM, 1,1.033854,237.500000,,5.76,176.3,T, , ,N,ARPA 1,T, *22<0x0D><0x0A>
<CORAL>--> 15:35:17 (COM6) $RATTM, 1,1.033854,237.500000,,5.76,176.3,T, , ,N,ARPA 1,T, *22<0x0D><0x0A>
```

rdouguet, 2022-12-15:
Thank for your comments Hakansv.

In fact, during my last test, I had no boat position but I thinking that it was no need to have position for the radar returns the RATTM frame (since the RATTM consists to send a bearing and distance between the radar and the target).

So, to have a valid position, I need to send GPRMC frame to OpenCPN ? Or I need to send other frames as the HDT ?

And I have another question : 
- The RATTM virtual frame is build by OpenCPN or it is returns directly by the radar when we enable ARPA mode in guard zone ?

Thanks for your help,

Hakansv, 2022-12-15:
Ronan..
1. All ARPA handling is made by the radar_pi plugins and has nothing to do with a connected radar scanner, as long we receive radar blobs.  
The NMEA0183 message RATTM is though made by the plugin and sent to OCPN core multiplexer.
2. You need a position and heading, HDT/HDG,  for chart overlay. If no heading is available you may for test purposes experiment to use COG instead. See radar preferences.

rdouguet, 2022-12-16:
Thanks for your feedback. 

In fact, I did not understand the ARPA mode on OpenCPN. As the HALO 20+ radar is able to detect up to 20 MARPA targets, I thought that OpenCPN just read the data provided by the radar and send them throught UDP (if enable) and display ARPA targets on the charts (if position is enabled)...

On my system, there is an IMU and a GPS => so I can build nmea frame ($GPRMC, $HDG) to generate the position and heading of vessel to OpenCPN. Next week, I will check if the ARPA target will be correctly generated...

I don't know why one time during a test, the frame "RATTM" has been correclty send whereas I have no position of boat... Maybe since before this, I have replayed data with VDR plugin and OpenCPN keep the last Heading...

I'll let you know if everything works next week.

Thanks,
Ronan.

rdouguet, 2023-01-03:
Hi All,

Sorry for my late answer but with the holidays I could not have access to the rib to do this test.
Normally, I will be able to test this solution later this week.

Thanks,
Ronan.

rdouguet, 2023-01-06:
Hi,

I tested yesterday this solution on the boat in a shed and the result is the same => I failed to generate an ARPA target.

Information about system :
- Radar Halo 20+
- OpenCPN in 5.6.2 (on Ubuntu), I also teste on windows PC with the same results
- Plugin Radar v5.3.4

GPRMC and AIHDT frames are sent to OpenCPN around 5 Hz.
The Position and HDT seem correctly received (see the following screenshot). I tested to activate the ARPA on the Guard Zone 1 or 2 but no improvement...

[image]

There are two month, I already tested on another PC and without send the RMC and HDT frames to OpenCPN and the ARPA sentences were correctly detected and send...

[image]

Do you have another advises or tests than I can be do ?

Thank for your help,
Ronan

douwefokkema, 2023-01-06:
The inner diameter of yr guard zone is larger than the outer diameter. So there are no targets in between. This is how it should be.
Douwe

rdouguet, 2023-01-06:
I have already tested to set the inner diameter smaller than largre diameter and the software crash... So I supposed that these parameters have been reversed. Moreover the color of guard zone are correctly display on radar view and this setting has already work (2.00 for inner and 0.00 for outer).

I can be retest this the next time to be sure...

douwefokkema, 2023-01-06:
But on your picture there are two contours of targets shown. Have you set View->AIS ARPA on PPI to On?

douwefokkema, 2023-01-06:
Still you will not have targets if inner range > outer range. If this gives a crash please report the crash, with all the initial conditions and a log file.
It does not make sense to have inner range = 0, one is not interested in the large central blob around the radar. May be that case (inner = 0) has been omitted in the tests.
You could also try to get a target by clicking the right mouse button on a blob on the overlay (set window overlay on), and select Aquire Marpa Target.

rdouguet, 2023-01-06:
Indeed, the inner range to 0 is maybe not take into account !

I just remember that we have some bugs (and software crashes) when we changed the range settings => That is why, at the end we let the inner > outer... but I agreen with you it does not make sense :)

I will check this the next week.

Thanks for your time.

rdouguet, 2023-01-12:
Douwe, thaks a lot for your advise since now all works :)

Indeed, the inner range must be above to outer range => but I have lot of crashes when I try to change these parameters when the radar is plug to the computer... I don't know why. My solution is to set this paramter before the radar is plugged.

I attach a opencpn log file that I had during a crash but I do not see error in this file :
[OpenCPN_crash.log](https://github.com/opencpn-radar-pi/radar_pi/files/10402653/OpenCPN_crash.log)

[image]

Otherwise, all is ok. The ARPA sentence are correctly send to my system even without HDT frame or GPS frame.

Now I can continue my sea trial :)

Thanks,
Ronan

douwefokkema, 2023-01-13:
Thanks for the log and screen image., this helps a lot. ARPA can not function properly without a heading and a position. In most cases it will produce useless results, it may even crash when the heading contains some random large value. To prevent this situation we block ARPA and MARPA in the coming release when there is no heading, this solution is already implemented in the master branch on Github. Until a new release is out, take care to use ARPA with a heading source and a position source. Just imagine, how can a system ever locate the position of a target without knowing its own position? But this is a bug that will be solved in the next release. Until then better provide the data required. And don't make the inner guard boundary zone 0, that makes no sense.

closed 2023-01-16 by rdouguet