opencpn-radar-pi/radar_pi #221: How to add my radar? How to normalize the collected data of radar to this plugin? I provide some radar acquisition data and radar data format.

cheungxiongwei, 2022-12-21:
### How to normalize the collected data of radar to this plugin? 

I provide some radar acquisition data and radar data format.

format description:
```
fixed size：2084 byte
single spoke format：
+------------+----------------+-------------+-----------------+--------------+
| DataHeader | BackEndDataLen | BackEndData | FrontEndDataLen | FrontEndData |
+------------+----------------+-------------+-----------------+--------------+
| 12 byte    | 2 byte         | 1234 byte   | 2 byte          | 834 byte     |
+------------+----------------+-------------+-----------------+--------------+

==================================================================================

fixed size：12 byte
DataHeader format：
+-------------+---------+---------+---------+--------+---------+--------+--------+
| HeaderMagic | Reserve | Reserve | Reserve | Signal | Azimuth | Sample | Tune   |
+------+------+---------+---------+---------+--------+---------+--------+--------+
| 0x55 | 0xAA | 0x01    | 0x01    | 0x01    | uint8  | uint16  | uint16 | uint16 |
+------+------+---------+---------+---------+--------+---------+--------+--------+

If the `Signal` value is not equal to 2, the data is discarded.

Azimuth range: 0~5399
Sample range: 0~1023
Tune range: 0~4095

==================================================================================

fixed size: 9872 bit (1234 byte * 8 bit)
Example BackEndData format:
+--------+--------+--------+--------+-----------+
| Data 1 | Data 2 | Data 3 | ...... | Data 1234 |
+--------+--------+--------+--------+-----------+
| 0x00   | 0x00   | 0x3F   | ...... | 0xE0      |
+--------+--------+--------+--------+-----------+

0x3F = 0011 1111
0xE0 = 1110 0000

Each bit in each byte represents a piece of data.
If it is equal to 1, it means that there is data. Otherwise, there is no data.

The actual physical distance per bit is 3.75 meters

1234 * 8 * 3.75 = 37020 meters

==================================================================================

fixed size: 834 byte
Example FrontEndData format:
+--------+--------+--------+--------+----------+
| Data 1 | Data 2 | Data 3 | ...... | Data 834 |
+--------+--------+--------+--------+----------+
| 0x00   | 0x00   | 0xFF   | ...... | 0xE0     |
+--------+--------+--------+--------+----------+

Each byte represents the strength of the echo data

0x00:Express data strength is the weakest
0xFF:Express data with the strongest strength
Data1 ~ Data475(475 byte):Indicates the current range

```
[3NM-Echo.zip](https://github.com/opencpn-radar-pi/radar_pi/files/10275856/3NM-Echo.zip)

keesverruijt, 2022-12-28:
Interesting, but your post raises more questions than it answers. What brand and type is this? Why "Frontend" and "Backend" data? What does this mean?

Anyway, since we don't have this radar, we will not do the integration work for you, you have to do this yourself. This has been described on the wiki here: https://github.com/opencpn-radar-pi/radar_pi/wiki/Adding-new-radar-types

cheungxiongwei, 2023-03-06:
Consulted with the radar manufacturer, the radar is Furuno model.

closed 2023-03-06 by cheungxiongwei