opencpn-radar-pi/radar_pi #222: Radar PI plugin for Raspberry Pi4B running OpenPlotter distro with OpenCPN

AlanJHancock, 2023-01-31:
The OpenCPN plugins page here https://opencpn.org/OpenCPN/plugins/radarPI.html
Indicates to use the 5.2 - Use the built-in Plugin Manager to install this plugin opencpn-plugin-radar
The plugin does not appear in the OpenCPN Plugin menu so assume the package for Debian has not been created.
Have tried to get it via flatpak both the recent version and the BR24 version with no success.
I would really like to have access to this plugin if possible to use my Lowrance 3G radar.
Any help to get this plugin working on Pi4B would be much appreciated.

AlanJHancock, 2023-02-04:
I was finally able to make a correct plugin tarball file that was accepted by the OpenCPN 5.2 Plugin Manager import plugin feature.
1. Clone the GIT sources using the CLI interface in a terminal window using gh command:
gh repo clone opencpn-radar-pi/radar_pi 
You may need to install gh command from [https://cli.github.com/] a personal key may need to be created for the download you will be prompted for that or use a web page login.
2. Extract the GIT plugin source file radar_pi-master.zip and enter the radar_pi directory in a terminal window
3. The OpenCPN Debian does not have cmake utility so install that for your particular distribution.
4. After cloning, in the radar_pi directory, setup the library submodule and enter a fresh working directory:
$ git submodule update --init opencpn-libs
$ rm -rf build; mkdir build; cd build
A "normal" (not flatpak) tar.gz tarball which can be used by the new plugin installer available from OpenCPN 5.2.0 is built using:
$ cmake ..
$ make tarball

closed 2023-02-04 by AlanJHancock

Hakansv, 2023-02-04:
Alan..
Yes, Radar_pi lacks support for some system in the pluginmanager. We also need to support next OpenCPN release 5.8 in a close future. 
Good you meanwhile managed to fulfill your needs.