opencpn-radar-pi/radar_pi #223: Garmin GMR 18 xHD - I can not power it up

pelakuti, 2023-03-07:
Hi, my name is Pelayo, from Spain and I am a cadet on this electronics world.
I have the dream of having my radar on OpenCPN from many years ago and I am right now on the way to try to achieve that goal thanks to the developers of this pluging. I am very grateful. Thank you.
I bought the GMR 18 xHD because of the price and because I dont have any MFC, just the computer, and I read on the wiki that this model would be fitting for this purpose.

I can not power it on and I want to tell you the steeps that I have followed in order to find the mistake:
Right now the antenna is connected to the 24V terminal block bus from my boat. I will take the dar home and connect directly to a battery to continue the testings.
I bought the famous Ampherol MRJ 570 but I did not solder the cables on it and I did not attach any switch (I know that on the wiki is written that this is compulsory) because I have seen pictures on the net from other users and they are all different :(
[image]
[image]
[image]
Wich one of the three is the correct? Once I know this for sure I will solder this and attach the switch.
After reading the wiki, in my opinion the connection should me made between this 2 pins:
[image]
Am I right? I proceed with the soldering and the switch?
Another question is if I have to solder any aditional cable and if I have to cenect it somwhere else? I know I am in the 2nd scenario (no MFD, directly to PC)  but in the picture of the scenrio 1, the ground symbol made me crazy. Do I have to connect the ethernet pin 5 to the battery? Is not just to connect pin 5 from the Amphenol to the "conector shell, chasis ground of the Amphenol" too?
Regarding to the ethernet cable that goes to the computer: does it has to be any kind of special apart from beeing cat 5 minimun? I am using a simple cat6e. Do I have to cut the 5 wire from this cable?

I am desperate, I tought that this would be easy but has become a nightmare to me. I could not power it on, the Status LED is all the time off. In all the forums I have read, I found people with the same issues all of them where not replayed. I have read all the tabs I found on google related to the pin 5 :(
I am praying  on my knees for your help and that maybe the origin is just in the non soldering.....
Apologize, is the first time I do this and I have no many idea.
Thank you very much and cheers for your amazing job!

keesverruijt, 2023-03-07:
Unfortunately the core guys don't have any actual Garmin experience, only Navico. Maybe @trudK45 can help out here.

pelakuti, 2023-03-07:
Thank you very much Kees. I am crossing my fingers awaiting for @trudK45 or someone else that can iluminate me.
💯 Thank you.

trudK45, 2023-03-08:
Be calm. I am curtain we will get your radar up an running soon.
I have a Garmin GMR 18 xHD since 2015 and it is still running fine.
  
If this is the Ampherol connector you have it is probably correct to connect the switch between the pins you have indicated.

[image]
To be sure you can check that pin 5 is the one we/you suspect by using an ohm meter and verify the connection between the marked pin 5 and pin 5 in the female connector on the Amphenol connector. The pinning in the connector is as in the following picture.

[image]

Apart from the cable from the radar dome you need a RJ45 cable to the computer.

bdbcat, 2023-03-08:
1. The photo #3 (with switch and yellow/red wires) is the original correct one, built by me, the original plugin developer.  It is known to work.
2. There is no need to connect any part of the connector/switch arrangement to ship's ground.  All is internal.
3. Standard Cat5/6 cable is fine.
4. A switch is necessary.  The scanner power must be applied before the switch is activated, to allow electronics startup time.

Good Luck
Dave

pelakuti, 2023-03-08:
Thank you, thank yo, thank you very much. 

Tomorrow I am going to the boat and testing all of this, I am thinking even to take the garmin to my house and one battery to continue with the testings at home.
Yes, @bdbcat I took the picture from one of your post at cruise forum, so I am ultra happy that you could reply me here :)
Definetly I will solder the cables and attach a switch.
My doubt now is that according to the Amphenol datasheet the pin 5 is in the upper line of pins and if my eyes dont lie to me, in your picture the cable is soldered to the "pin 4", in the lower line of pins. Am I correct?

Thank you.
Pelayo.

keesverruijt, 2023-03-08:
Also consider that maybe the difference in pin use can be explained by putting the radar in the connector on the other side.

trudK45, 2023-03-08:
> My doubt now is that according to the Amphenol datasheet the pin 5 is in the upper line of pins and if my eyes dont lie to me, in your picture the cable is soldered to the "pin 4", in the lower line of pins. Am I correct? 
> 
I think the reason for the confusion is due to difference in orientation and difference in the layout.
I am a bit surprised locking at the third of your pictures. Are the connectors the same and one of them not "upside down" ?
If the connectors are identical I would expect there to be a crossover in the PCB to get each pin between the 2 connectors connected to the same pin number.
Are the 2 connectors identical ?
To be sure verify that pin 5 on both connectors is connected to the pin marked 5 on the back of the PCB. Then you can go ahead and solder in the switch.

pelakuti, 2023-03-08:
Hi, thank you all for your replies.

I will attach here some pictures of the ethernet cable that came in the box of the radar. In my opinion, this is a crossover:
[image]
I dont know if older radar work with straight ethernet cables and now they changed to a crossover.

Then I took some pics of both sides of the Ampherol, labeling both sides pins following the same pattern that Amphenol writtes on the datasheet:
[image]
Here all looks straight, like if you cut the cables and then join them exactly on the same position.
But, I noticed that when the people label the pin numbers on FEMALE conectors, the order of the pins from the right to the left. Exactly the oposite way labeled by Ampherol:
[image]

So now I am even more lost. If all the above is correct, Has to be a croosover cable the one that goes to the PC? 

Thank you very much.

pelakuti, 2023-03-08:
I found this than can be also very interesting right now. I think the supplied cable to me is propietary from Garmin:
https://support.garmin.com/en-US/?faq=MNUtEt8k5EA2s8dR9eqIQ9
https://static.garmin.com/pumac/Marine_Ethernet.pdf

Hakansv, 2023-03-08:
Configuration A or B is to be followed when you make a new cable yourself. A or B doesn't matter as long you use the same in both ends. (Has nothing to do with crossover or not.)
Use the delivered cable. Locate pin 5 on the cable plug. Make sure you solder so it's connected to pin 5 on the plug when plugged.

keesverruijt, 2023-03-09:
I wouldn’t worry so much and just try it out. 4 and 5 are both the center pair, not used for 100mbit. So just try both sides. 

Ethernet is extremely robust.

pelakuti, 2023-03-09:
Excellent. As soon as I have a free moment I will make the soldering and tell you all the progress.

Thank you all once again for your incredible help.

pelakuti, 2023-03-10:
Hello to all! I am back with good news!!
I soldered a switch as suggested by bdcat and is working!!
I am the happiest man in the world and is thank to you all guys. Incredible! I have no words, without your help this would not be possible.
If any of you comes to Mallorca dont doubt to contact me because I have to invite you to whatever you want.
So, thank you once again guys!

trudK45, 2023-03-10:
> Hello to all! I am back with good news!!

I will keep this open until you have got a radar picture within OpenCPN.
You might still need some guidance to get it working.

pelakuti, 2023-03-10:
Excellent. I'm mounting it on Monday onboard so this day I will take some pics and upload here. Thank you very very much Trudk45!

Hakansv, 2023-03-22:
@pelakuti 
Time to close this?

closed 2023-03-22 by Hakansv