opencpn-radar-pi/radar_pi #224: AIS targets to PPI window is broken to OCPN 5.7.x

Hakansv, 2023-03-22:
Seems there may be some changed API for AIS targets to PPI in 5.7.x beta?
The Json transfer still works though.

Hakansv, 2023-04-02:
Running on Win10 Debug in VS.
O 5.7.1 present master. Radar_pi present master.
I can brake on "PlugInAISDrawGL()" in RadarCanvas.cpp but don't understand where to look for possible cached AIS targets.

[image]

douwefokkema, 2023-04-02:
I also tested, first I had a crash in PlugInAISDraw(), I pulled the latest version of O, no crash anymore. It seems problem is in PlugInAISDraw(). O itself is showing AIS fine. But radar_pi is using TTM with a different symbol.

douwefokkema, 2023-04-02:
The TTM symbol is showing fine on the overlay.

Hakansv, 2023-04-02:
"TTM symbol on overlay.."
Is it the ARPA on chart overlay you mean?. Yes that's fine.

[image]

Hakansv, 2023-04-02:
Douwe..
> "But radar_pi is using TTM with a different symbol."

 Where, except sending ARPA info to "PushNMEABuffer", is TTM used in radar_pi?

douwefokkema, 2023-04-02:
Only for ARPA.
Other thing: is the chart_scale = 0 OK in yr debug output? looks strange. I will check tomorrow.

Hakansv, 2023-04-02:
If I understand this correct PlugInAISDrawGL in RadarCanvas.cpp will send data to Pluginmanager.cpp that will use the data and 
call AISDraw() to find the PPI window and draw AIS targets on that screen position?
So far I can follow it. But I'm not sure what content would be there and what's missed since no of the AIS targets found are drawn on the PPI window. I may need help to exterminate.
[image]

keesverruijt, 2023-04-02:
Funny, I just started on this this weekend as well. I haven't gotten a full O dev env where I can trace into O yet, working on that. I've looked at O `viewport.cpp` and I don't think `chart_scale` matters, but setting it to a non-zero value will not hurt. But maybe some other viewport variable is now needed.

keesverruijt, 2023-04-02:
Possibly lat_min/lat_max/lon_min/lon_max or bValid. Setting them certainly wouldn't be bad.

bdbcat, 2023-04-03:
Hakan...
Not sure I am following this.  
Can you re-state the problem, functionally?

LeaseOnLife, 2023-04-03:
Not Hakan here, but the original issue was about the AIS targets not showing on the PPI window, they did in the past and there is an option to toggle them on/off on that PPI window

[image]

bdbcat, 2023-04-04:
I have reproduced the error, working on a fix now.  Not a flaw in radar_pi.  It is regressive change in OCPN core involving OpenGL upgrades recently implemented.
More as I proceed.

Hakansv, 2023-04-05:
Works fine, very good!
Thanks

closed 2023-04-05 by Hakansv

keesverruijt, 2023-04-06:
Confirmed fixed with https://github.com/OpenCPN/OpenCPN/commit/0d019bc72003e01863791f30c5abdea5a1294cd6