opencpn-radar-pi/radar_pi #225: latest libwx3.2 update breaks radar plugin

rahra, 2023-04-08:
I was using the radar plugin successfully until the latest libwx3.2 update which occurred on 6th of April.
I'm using Ubuntu Linux 22.04 (Jammy) having OpenCPN installed from the Ubuntu OpenCPN PPA and the radar plugin from the beta catalog (since there is no version in the master). The Ubuntu ppa also delivers the libwx libraries.
Everything is a straight forward standard installation without any special or unusual installations.
Today I did an apt upgrade and now the radar plugin gets disable because "there is some problem" according to OpenCPN.
The dpkg.log shows that the libwx3.2.... libraries have been updated.
Unfortunately I could not find out yet which version was installed before because I would love to downgrade these.

douwefokkema, 2023-04-08:
You are not explicit in what versions you were using when the problem occurred. But I assume you were using O5.6.2 as that is the one from the PPA. The beta plugin is the version for the new O5.7. These are indeed not compatible, so there is nothing wrong.

rahra, 2023-04-08:
It works with the jammy-backport which is OpenCPN 5.6.2 and I installed the radarplugin found in debian-wx32-x86_64:11 which is 5.5.0-beta.
In the Ubuntu PPA OpenCPN there is the version 5.7.5. I think I installed the radar plugin from the according Ubuntu catalog. I will reinstall everything and try with the Debian version of the plugin.

Hakansv, 2023-04-08:
For info is OCPN v 5.8 to be released the coming days and the available radar_pi v5.5.0 beta will fit there.

keesverruijt, 2023-04-08:
Debian built plugins may be not compatible with Ubuntu because they use (slightly) different wxWidgets libraries. This is one of the reasons OpenCPN on Linux is moving towards Flatpak builds, this forms a single well defined Linux target so we only need to build for a single x86_64 target.

So unless you produce a reproducible case where you say download this particular O version and then install a plug-in from its master or beta catalog that shows up in the gui, and this then doesn’t work, there is no bug here. And if there is it is likely more a bug in OpenCPN target matching code than the plug-in. Manually importing a .tar.gz file doesn’t count.

If you do find a bug, we need you to supply exact version numbers and full logs. There is no way we can guess what you are doing from the multitude of combinations possible on Linux.

bdbcat, 2023-04-08:
Kees....
This is a little strange.
I just tested, latest (preliminary Release) from OpenCPN Production PPA, ubuntu 22.04.  This does indeed use a recent rebuild of wx32, using latest upstream sources, created by leamas.
And, sure enough, Radar_pi v 2.2.0-beta (from beta catalog) fails to load.
Missing symbol (de-mangled) is wxGLCanvas::IsShownOnScreenEv()

So, we can point to the update of wx32.  Need input from leamas here, I think.
the exact log message is:

`10:31:04.105   ERROR dlunix.cpp:120 Failed to load shared library '/home/dsr/.local/lib/opencpn/libradar_pi.so': /home/dsr/.local/lib/opencpn/libradar_pi.so: undefined symbol: _ZNK13wxGLCanvasEGL15IsShownOnScreenEv, version WXU_3.2
`

nohal, 2023-04-08:
The problem is this:
```
11:34:11.744   ERROR dlunix.cpp:120 Failed to load shared library '/home/nohal/.local/lib/opencpn/libradar_pi.so': /home/nohal/.local/lib/opencpn/libradar_pi.so: undefined symbol: _ZNK13wxGLCanvasEGL15IsShownOnScreenEv, version WXU_3.2
```
The target seems to be selected correctly by the core app (`ubuntu-wx32-x86_64:22.04) and pulls the correct `debian-wx32.wx32-11-x86_64` tarball far the plugin.

@leamas, rings any bells? FWIW, my plugins from the beta catalog all seem to work fine with this build.

keesverruijt, 2023-04-08:
Aha. I just tested using a fresh VM with Ubuntu 22.04,  `ppa:bdbcat/opencpn` and that works fine. Apparently that is not using the updated wx libraries yet?

nohal, 2023-04-08:
Nope, the new build is just in `ppa:opencpn/opencpn`

keesverruijt, 2023-04-08:
Maddingly the plug-in source doesn't even call that IsShownOnCanvasEv() function directly.

nohal, 2023-04-08:
Maybe just a simple rebuild will solve it?

keesverruijt, 2023-04-08:
Are the shipdriver templates updated? This is where we get our definitions of what wx to pull from. I'll check that right now.

bdbcat, 2023-04-08:
I guess I should delete the new wx32 build from ppa:opencpn/opencpn, and pick up the wx dependency for the OCPN build to the ppa:bdbcat/opencpn.
@nohal:  Any other ideas?

nohal, 2023-04-08:
I will try to install the wx packages from `ppa:bdbcat/opencpn` manually and see what happens

nohal, 2023-04-08:
The problem likely will be that the ones in your repo are `3.2.1+dfsg-1ubuntu1~bpo22.04.2`, while the ones pushed to `ppa:opencpn/opencpn` are `3.2.2+dfsg-1ubuntu1~bpo22.04.1` and the ABI probably broke between them...

keesverruijt, 2023-04-08:
Well, at least we should have a single source of truth for which wxWidgets version we build both O and plugins from. I've now updated the plugin to sd3.2.3 template files, and will check whether that changed anything to the build.

nohal, 2023-04-08:
The CI builds for the plugins (At least shipdriver, radar and mine) build against 3.2.1 (https://github.com/opencpn-radar-pi/radar_pi/blob/master/ci/circleci-build-debian.sh#L37)

bdbcat, 2023-04-08:
kees...
Please consider that "check" to be for investigation only.  If possible, we should not break all existing plugins.  Probably means that we will want to revert to prior wx32 build in my beta PPA.

keesverruijt, 2023-04-08:
From the wxwidgets.org main site: `This release is API and ABI-compatible with the previous 3.2.x releases, so the existing applications don’t even need to be rebuilt to profit from all the fixes above if they use shared/dynamic libraries. And if they do need to be recompiled, this can be done without any changes to the code.` Sigh, apparently not for our use case.

keesverruijt, 2023-04-08:
Let's discuss on Zulip?

bdbcat, 2023-04-08:
OK.  Easier.

nohal, 2023-04-08:
> OK. Easier. What topic?

https://opencpn.zulipchat.com/#narrow/stream/332168-Master---5.2E8.2E0--.28was.3A-comms.29/topic/wx3.2E2.2E2.20compatibility

keesverruijt, 2023-04-09:
A quick update: This is a real mess, caused by the fact that Debian and Ubuntu broke the ABI between wx 3.2.1 and 3.2.2 by compiling wxWidgets with a different set of options. 

We have concluded that we are better off by using wx 3.2.2 for the platform where we have "home made wxWidget libraries", e.g. Debian 11 and Ubuntu 22.04. It should fix itself on Debian 13 and Ubuntu 23.04.

This does mean that for those platforms we need to rebuild the radar plugin, which will then only be compatible with OpenCPN releases using wx 3.2.2 -- like the one in the production opencpn PPA.

keesverruijt, 2023-04-10:
@rahra , can you try the latest tarball in https://cloudsmith.io/~opencpn-radar-pi/repos/opencpn-radar-pi-unstable/packages/ ? This should work.

keesverruijt, 2023-04-11:
Yes, you caught us uploading a broken macOS build to that repo. The CI process, for some reason, generates an incorrect SHA sum, on macOS only...

keesverruijt, 2023-04-11:
Yes, you caught us uploading a broken macOS build to that repo. The CI process, for some reason, generates an incorrect SHA sum, on macOS and some Linux only...

keesverruijt, 2023-04-12:
@rahra Thanks again, sorry we were a bit dismissive at first. This turned out to be a real mess.

closed 2023-04-12 by keesverruijt

rahra, 2023-04-13:
I know how this works, I'm open source developer since tens of years ;)

rahra, 2023-05-07:
Just to report back from my boat computer: works as expected.
Used versions: Ubuntu Jammy 22.04.2, OpenCPN 5.8.2 from official Ubuntu launch pad, flavor "jammy", Radar plugin 5.5.0 from OpenCPN catalog.