opencpn-radar-pi/radar_pi #226: Trouble getting radar working correctly

dougbrouwer, 2023-04-10:
Radar_PI 5.3.4, OCPN 5.6.2. Halo 20+. MacOS 12.4 is the only display.
I've set up a "dry run" by just setting the dome on top of the cabin, to prove to myself that the Halo+ will do what I want. I'm hoping to not have to mount the dome up the mast until I get everything working. 
I've searched both this forum and the cruisers forum to no avail, so I hope this isn't covered somewhere I haven't found.
I've got the radar configured to the point that it displays and/or overlays. My issue is that it only displays briefly, then goes to standby or "No radar", then cycles again to display briefly.
Here's a link to an online video (recordit.co) that shows what's happening in OpenCPN: [https://recordit.co/uFCP8hjoc3](url)

If at all relevant: Cortex M1 AIS, Furuno SCX-20 satellite compass, YachtDevices ethernet to NMEA 2000 converter, Netgear GS308E switch, Teltonika RUT240 router

Any ideas as to why the radar connection would be so intermittent?

Hakansv, 2023-04-10:
It seems being a network trouble. 
You mentioned a switch and a router. Pls describe how the Halo is connected to the PC. 
If possible try to connect the Halo direct to the PC's IP-port. 
Now I'm not a MacOs user but if a firewall is involved try to disable that for a test.

keesverruijt, 2023-04-10:
Let me guess, you are using WiFi to connect the Mac to the router?

dougbrouwer, 2023-04-10:
Bingo! Gave the Mac an ethernet connection and problem solved! Steady and continuous sweeps. Thanks very much to keesverruijt.

closed 2023-04-11 by dougbrouwer