opencpn-radar-pi/radar_pi #231: Flatpak build is rebuilding files

keesverruijt, 2023-04-16:
The flatpak build seems to run commands twice. 

For instance, it is running the NMEA0183 library link twice. This is most obvious when the `cmake` build system is changed to `cmake-ninja` in `flatpak/org.opencpn.OpenCPN.Plugin.radar.yaml`:

    [50/57] Linking CXX static library libs/nmea0183/libNMEA0183.a
    [54/57] Linking CXX static library opencpn-libs/wxJSON/libWXJSON.a
    [12/21] Linking CXX static library libs/nmea0183/libNMEA0183.a
    [17/21] Linking CXX static library opencpn-libs/wxJSON/libWXJSON.a

leamas, 2023-04-17:
Yes, I can reproduce this also on plain shipdriver builds. Digging

leamas, 2023-04-17:
This is indeed the core problem. A really nice catch!

However, just using BUILD_TYPE=flatpak in the manifest does not work in the general case. I'm on it, but probably need a night's sleep to figure out what to do. Perhaps even two...

keesverruijt, 2023-04-17:
Maybe change the flatpak instructions (yaml) to build type ‘simple’ then just call ‘make’ instead of cmake?

keesverruijt, 2023-04-17:
Hmm come to think of it, I think the best approach is probably to get rid of the _outer_ cmake layer, which in the runs a single generated script. 

Instead, have circle-build-flatland.sh generate a build-flatpak/build.sh script directly and feed that to flatpak builder using the ‘simple’ buildsystem. The mechanism to generate it could resemble the one used for cloudsmith sync so use an .in file where variables are substituted etc.

The cmake files can then remove most references to flatpak; it just becomes a tarball build.