opencpn-radar-pi/radar_pi #232: HALO Unable to receive data

liuchao1, 2023-04-17:
The radar can initially receive data, but after running for a period of time, it can no longer receive data and will continue to display Spinning up

liuchao1, 2023-04-17:
The radar is connected via wired.

keesverruijt, 2023-04-17:
I assume you are running opencpn 5.8 and plugin 5.5.0. If not  please upgrade, if yes please upload your logfile.

keesverruijt, 2023-04-17:
On windows disable the firewall, for testing.

keesverruijt, 2023-04-17:
And make sure wireless network is not connected to the same network as the radar. Safest is to switch it off for testing.

liuchao1, 2023-04-17:
I will try upgrade to opencpn 5.8 and plugin 5.5.0 ,for testing

liuchao1, 2023-04-17:
Thanks! The problem has been resolved !  The radar is locked .  Another question, will using the OpenCPN radar be locked?

closed 2023-04-18 by liuchao1