opencpn-radar-pi/radar_pi #233: Radar screen looses boat position

rahra, 2023-05-07:
The center of radar screen usually should be the boat position. When I move the mouse cursor over the chart, a cursor is displayed in the radar screen as well (small plus). If you hover over the boat in the chart the radar cursor is in the center. That is what is expected.

Occasionally I observed that somehow the radar screen looses this center. which means the boat moves but the center position of the radar screen is assumed to be somewhere else (on a past track point) although the radar picture itself obviously reflects what it actually sees.

I could not yet determine what exactly causes that, so what causes that issue.

This is meant to be a first report on that observation.

douwefokkema, 2023-05-07:
You can drag the radar window off-center with the mouse. After having moved the radar window, you can zoom in to see details on the image. Zooming in after moving does not change the range of the radar.

rahra, 2023-05-08:
The following screenshot shows what I mean. The mouse cursor was hovered exactly above the boat (cursor not visible in the screenshot) but within the radar screen this is off center.
[image]
Instead the center is where the last track segment seems to start in this example:
[image]

douwefokkema, 2023-05-08:
Can you get me a picture of the Information window (Window->Info...)?

keesverruijt, 2023-05-08:
> This checkbox seems also be necessary for MARPA targets. 

This should not be the case. It looks as if your heading data is not "understood" by the plugin, which is why Douwe asked for the Information Window content (I think.)

> With this checkbox disabled it does acquire any random targets on the screen and not the one where a place the cursor to within the radar screen. Not sure if this is the same issue or somehow related.

Sounds like it.

Are you using O 5.8.2 and the NMEA2000 interface of OpenCPN, or a NMEA0183 input in OpenCPN (via, for instance, Signal K or a N2K -> N0183 converter)?

Hakansv, 2023-05-08:
> I can read the correct magnet heading

You do need a true heading. If your sensor only serves a magnetic heading you may need to enable the WMM plugin?

rahra, 2023-05-08:
@keesverruijt I activated logging an I will observe it. Now on 1st watch, plenty of time ;) 
The installation is pure N2K, Actisense NGT-1 bridge to USB. All data is collected by SignalK and OpenCPN interfaces to SignalK with SignalK protocol on TCP port 3000. OpenCPN is 5.8.2 with Radar plugin v5.5.0. Every piece of software is up to date.

@Hakansv I activated the plugin, target acquisition worked immediately but I will observe this as well.

Hakansv, 2023-05-08:
Good. And please disable "enable COG as heading" if still left.

douwefokkema, 2023-05-09:
I do not yet completely understand your setup. Is NGT-1 connected to and handled by OpenCPN directly or to a SignalK server?

douwefokkema, 2023-05-09:
I just did a test with a N2k heading sensor connected to a NGT-1 to OpenCPN. Radar Info window shows heading Magnetic 0, that is wrong, so does the dashboard in O. Also radar shows a magnetic heading + WMM but no true heading, this is also wrong. So your observations are OK, and the new nmea2000 software needs some corrections, both in O and radar.
In a test plugin I can read the heading from the NGT-1 through O, no problem with my setup.

douwefokkema, 2023-05-09:
It seems to me that the function ParseN2kPGN127250() does not function properly, I will investigate further tomorrow. When I directly fetch the heading from the message payload, the heading is OK.

Hakansv, 2023-05-09:
About true, HDT, and magnetic, HDM, headings.
OCPN:
O use a HDT if received. If several are received the prioritization logic and its user setting controls whom to use. See [here ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:connections:priority)for info about how to set the priority. (O >= v5.8.0)
If only HDM is received and no variation is received O will use WMM to calculate a HDT. The same prioritization logic will control which HDM to use if more than one are present.
Note: For N0183 and N2k the prioritization logic can control which message/PGN -source to use. For SignalK, SK, O can just decide to use SK as such. For any message priority from SK to O, SK has to do all prioritization or selection by itself. 

Radar plugin:
The HDT used by O is also transferred to plugins as radar_pi. 
Radar_pi will use the HDT from O if no HDT is present from the scanner and it's not disabled in preferences. ([ ]"Ignore radar heading")

Dashboard:
D will if in first hand use the heading transferred from O itself as of the priority logic described above. 
(Douwe: PGN 127250 is not parsed by D. Only by O itself.)

Now for this particular case/issue:
Since @rahra stated: 

> All data is collected by SignalK and OpenCPN interfaces to SignalK with SignalK protocol on TCP port 3000.

all data received by OCPN are then controlled by SK. So any selection and priority has to be done in SK. No PGNs are parsed by OCPN in this case.
O can receive both HDT and HDM from SK if present. If only HDM is received WMM will be used as described.

Hakansv, 2023-05-09:
> I just did a test with a N2k heading sensor connected to a NGT-1 to OpenCPN. Radar Info window shows heading Magnetic 0, that is wrong, so does the dashboard in O. Also radar shows a magnetic [...]

Not using NGT-1 but YDNU-02. No reason for difference I suppose.
On my test all seems OK with N2k heading. On my picture both HDT and HDM are present from PGN127250. O performs a test for valid data before use. If the data is "0" it's though valid.
[image]

rahra, 2023-05-09:
> I do not yet completely understand your setup. Is NGT-1 connected to and handled by OpenCPN directly or to a SignalK server?

N2K bus <--> NGT-1 <--> SignalK-Server <--> TCP/SignalK-Protocol <--> OpenCPN

[image]

rahra, 2023-05-09:
To me this looks like the radar is showing true heading:

[image]

rahra, 2023-05-09:
A played the last night a lot and will do the same today. For me it looks like everything works fine since I enabled to WMM plugin.

Hakansv, 2023-05-09:
Very good! Pls excuse our possible interesting but maybe not relevant sidesteps.
When you're satisfied pls close this issue.
Thanks

rahra, 2023-05-09:
Issue seems to be solved.
The solution was to activate the WMM plugin.

closed 2023-05-09 by rahra