opencpn-radar-pi/radar_pi #234: Picture not updating in time when in docked position

aviatorhh, 2023-05-09:
When running the radar plugin in a separate window (floating) the radar image is running as it should and is showing live data. When in docked position the image pauses and only updates occasionally.
Plotter: OpenCPN 5.8.2, wxWidgets 3.0
Version: 5.5.0 (from OpenCPN repository)
System: RPI4

The picture below shows the discrepancy.
On the left overlay the port side red arc nearly is at 90° but on the docked display it is just at 45°.
The docked display is updated randomly but very slow/delayed.
 
[image]

By moving the divider bar a bit I can force a redrawing. Is this an OpenGL or wxWidgets issue? But why then the floating picture ist ok?

[image]

Here the overlay and the floating window are in sync.

keesverruijt, 2023-05-11:
I have similar issues but not identical:

- RPI4 with flatpak O 5.8.2 + wx 3.2 (current prod release) & catalog plugin 5.5.0: total freeze when I dock the pane.
- Ubuntu 22.04 aarch64 with PPA O 5.8.2 (also wx 3.2 base) & catalog plugin 5.5.0: works fine.

You say "wx 3.0". Are you sure? O 5.8.2 usually is built with wx 3.2, not 3.0. Is this your own compilation?

aviatorhh, 2023-05-11:
Yes, I compiled it myself. Anyhow, I had this freezing behavior before on 5.6 (orig) also.
I did wx3.2 (3.2.2.1) but had issues with the package plugin not being able to install any plugins from the repo (which wx3.2 is supported?). Trying to compile some plugins failed with cmake not catching/finding the local wx3.2 installation (tried a lot). Some other plugins worked fine as it seems they followed the design pattern (e.g. downloading opencpn-libs with git).
I would give it a go with 3.2 if I knew the best version (3.2.x?). But as you are running wx3.2 having the same issues I am not convinced.

keesverruijt, 2023-05-11:
What you do for yourself is up to you, of course. We use the `ci` scripts to build nowadays, this makes sure you get the same version as the current OpenCPN build. 

See, for instance `ci/circleci-build-debian.sh` function `install_wx32`.

aviatorhh, 2023-05-11:
Most plugins suggest their build process within their READMEs or similar (cmake - make). If I use for example ci/circleci-build-debian.sh for raymarine_autopilot_pi I get
```
-- PluginConfigure: wxWidgets components: base;core;net;xml;html;adv;aui;gl
CMake Error at /usr/local/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS)
```
And that is where it always stops but not on every plugin. radar_pi is fine. It fails on raymarine_autopilot_pi and NmeaConverter_pi. I have searched and tried almost everything so I stopped wasting time and went back to a running config (wx3.0).

But to come back on the topic I do not see that as a problem as you are having same issues on the display.

douwefokkema, 2023-05-11:
Just a remark on your compile problems. In my case these issues were solved as follows. Get the environment right. First compile OpenCPN itself: clone O, and run OpenCPN/ci/generic-build-debian.sh. After that the plugins are just a piece of cake.

aviatorhh, 2023-05-11:
Yes, hat does work on some. But could you do me a favour and try this on https://github.com/RooieDirk/NmeaConverter_pi ?Thanks!

> Just a remark on your compile problems. In my case these issues were solved as follows. Get the environment right. First compile OpenCPN itself: clone O, and run OpenCPN/ci/generic-build-debian.sh. After that the plugins are just a piece of cake.

aviatorhh, 2023-05-11:
I now have done a fresh O build against wx3.2.2.1 again and done the radar plugin (git clone) just now (Radar-5.5.1-alpha+2305111729.5729cec_debian-11-arm64.tar.gz).
With this setup it is running ok now!! No freeze while docked. So wx3.2 solved it?
But still got the compiling issues with NmeaConverter_pi and raymarine_autopilot_pi. At least the NmeaConverter is needed.

aviatorhh, 2023-05-16:
I may have found something else. I am not sure if this is only on my platform (RPi4, Debian 11).
When I close O with the radar plugin in docked setup and then start O again it crashes after start while trying to display the radar. Shall I provide some more information here or open a new issue?

keesverruijt, 2023-05-17:
No I think this is related. We use an old workaround to fix AUI problems where on shutdown the code notes the docked state, undocks the pane and then quits. On startup it notes the “docked” preference and then creates then docks the pane. If this crashes then it is certainly in the same area of code!

aviatorhh, 2023-05-17:
Ok. Actually all I can see is it ends with a Segment Fault. What can I do to debug it more deeply? BTW, my workaround is a script that hides the radar on startup by manipulating the opencpn.conf prior start. After O has come up I can activate the docked radar without problems manually.

keesverruijt, 2023-07-11:
Build everything in Debug mode and start OpenCPN under gdb.