opencpn-radar-pi/radar_pi #235: The plugin is not compatible with this version of OpenCPN, and will be uninstalled

VLongobardi, 2023-05-10:
Hi all,

First of all, thanks for your work.

I have some problem with the plugin installation.
I compiled the code just after cloned git, following the instruction inside "INSTALL.md"
I tried to compile both latest version (5.5.1 alpha) and the one that I installed with OpenCPN plugin manager and I'm currently using (5.5.0).
In both cases I get the same error when I try to install the plugin: "The plugin is not compatible with this version of OpenCPN, and will be uninstalled ."

My OpenCPN version is 5.8.2

That is the log:

```
17:37:28.980 MESSAGE OCPNPlatform.cpp:890 Getting DefaultSystemLocale...
17:37:35.074 MESSAGE update_mgr.cpp:218 Selected update: Radar
17:37:35.077 MESSAGE pluginmanager.cpp:572 Installing Radar
17:37:35.085 MESSAGE plugin_handler.cpp:1246 Installing Radar-v5.5.0_ubuntu-gtk3-20.04-x86_64.tar.gz from local cache
17:37:35.220 MESSAGE plugin_loader.cpp:1124     Plugin "/home/pc-username/.local/lib/opencpn/libradar_pi.so" depends on library "libwx_baseu-3.2.so.0", but the main module was built for "libwx_baseu-3.0.so.0".
17:37:35.220 MESSAGE plugin_loader.cpp:1143 Plugin is compatible by elf library scan: false
```

Do you have any suggest? 
Thank you

keesverruijt, 2023-05-10:
As the error message says, update your wxWidgets version to 3.2

keesverruijt, 2023-05-11:
The compilation instructions are somewhat outdated. Your best bet is to follow our continuous integration scripts, these always work /on the right platform/. For instance, either run or investigate `ci/circleci-build-debian.sh` for your platform.

keesverruijt, 2023-05-12:
Make sure you set

    export BUILD_WX32=true

before calling the ci script to use/download wx 3.2.

keesverruijt, 2023-05-12:
See `.circleci/config.yml`.

VLongobardi, 2023-05-26:
Hi,

VLongobardi, 2023-05-26:
Hu

VLongobardi, 2023-05-26:
Hi, 
as you suggested, I we ran and investigated `ci/circleci-build-debian.sh` and before ran we set `export BUILD_WX32=true`.
When I ran the `ci/circleci-build-debian.sh` I got an error from this line:
```
sudo sed -i '/^user_mask_fits/s|{.*}|{ /bin/true; }|' \
    /usr/lib/$(uname -m)-linux-gnu/wx/config/gtk3-unicode-3.2
```
I investigated and commented the line and this time I got no error. The output of the ran is:
```
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/nb_NO/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/nl_NL/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/pl_PL/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/pt_BR/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/pt_PT/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/ro_RO/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/ru_RU/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/sv_SE/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/th_TH/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/tr_TR/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/vi_VN/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/zh_CN/LC_MESSAGES/opencpn-radar_pi.mo
-- Installing: /home/user/Desktop/radar_pi/build-/app/files/share/locale/zh_TW/LC_MESSAGES/opencpn-radar_pi.mo
make[3]: Leaving directory '/home/user/Desktop/radar_pi/build-'
Built target tarball-install
make -f CMakeFiles/tarball-finish.dir/build.make CMakeFiles/tarball-finish.dir/depend
make[3]: Entering directory '/home/user/Desktop/radar_pi/build-'
cd /home/user/Desktop/radar_pi/build- && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/user/Desktop/radar_pi /home/user/Desktop/radar_pi /home/user/Desktop/radar_pi/build- /home/user/Desktop/radar_pi/build- /home/user/Desktop/radar_pi/build-/CMakeFiles/tarball-finish.dir/DependInfo.cmake --color=
Dependee "/home/user/Desktop/radar_pi/build-/CMakeFiles/tarball-finish.dir/DependInfo.cmake" is newer than depender "/home/user/Desktop/radar_pi/build-/CMakeFiles/tarball-finish.dir/depend.internal".
Dependee "/home/user/Desktop/radar_pi/build-/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/user/Desktop/radar_pi/build-/CMakeFiles/tarball-finish.dir/depend.internal".
Scanning dependencies of target tarball-finish
make[3]: Leaving directory '/home/user/Desktop/radar_pi/build-'
make -f CMakeFiles/tarball-finish.dir/build.make CMakeFiles/tarball-finish.dir/build
make[3]: Entering directory '/home/user/Desktop/radar_pi/build-'
cmake -P /home/user/Desktop/radar_pi/build-/finish_tarball.cmake
-- Creating tarball Radar-5.5.1-alpha+2305250807.5729cec_ubuntu-gtk3-20.04-x86_64.tar.gz
-- Computing checksum in Radar-5.5-ubuntu-gtk3-20.04.xml
make[3]: Leaving directory '/home/user/Desktop/radar_pi/build-'
Built target tarball-finish
make -f CMakeFiles/tarball.dir/build.make CMakeFiles/tarball.dir/depend
make[3]: Entering directory '/home/user/Desktop/radar_pi/build-'
cd /home/user/Desktop/radar_pi/build- && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/user/Desktop/radar_pi /home/user/Desktop/radar_pi /home/user/Desktop/radar_pi/build- /home/user/Desktop/radar_pi/build- /home/user/Desktop/radar_pi/build-/CMakeFiles/tarball.dir/DependInfo.cmake --color=
Dependee "/home/user/Desktop/radar_pi/build-/CMakeFiles/tarball.dir/DependInfo.cmake" is newer than depender "/home/user/Desktop/radar_pi/build-/CMakeFiles/tarball.dir/depend.internal".
Dependee "/home/user/Desktop/radar_pi/build-/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/user/Desktop/radar_pi/build-/CMakeFiles/tarball.dir/depend.internal".
Scanning dependencies of target tarball
make[3]: Leaving directory '/home/user/Desktop/radar_pi/build-'
make -f CMakeFiles/tarball.dir/build.make CMakeFiles/tarball.dir/build
make[3]: Entering directory '/home/user/Desktop/radar_pi/build-'
make[3]: Nothing to be done for 'CMakeFiles/tarball.dir/build'.
make[3]: Leaving directory '/home/user/Desktop/radar_pi/build-'
Built target tarball
make[2]: Leaving directory '/home/user/Desktop/radar_pi/build-'
/usr/bin/cmake -E cmake_progress_start /home/user/Desktop/radar_pi/build-/CMakeFiles 0
make[1]: Leaving directory '/home/user/Desktop/radar_pi/build-'
+ ldd app/Radar-5.5-ubuntu-gtk3-20.04/lib/opencpn/libradar_pi.so
	linux-vdso.so.1 (0x00007ffffbfb7000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007fb4e2a52000)
	libwx_baseu-3.0.so.0 => /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0 (0x00007fb4e278e000)
	libwx_gtk3u_core-3.0.so.0 => /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0 (0x00007fb4e210a000)
	libwx_gtk3u_aui-3.0.so.0 => /lib/x86_64-linux-gnu/libwx_gtk3u_aui-3.0.so.0 (0x00007fb4e206b000)
	libwx_gtk3u_gl-3.0.so.0 => /lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0 (0x00007fb4e2051000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fb4e1e6f000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fb4e1d1e000)
[... cut ...]
07:55:49.747 MESSAGE OCPNPlatform.cpp:1044 Loaded translation catalog for:  opencpn-grib_pi
07:55:49.747 MESSAGE OCPNPlatform.cpp:1044 Loaded translation catalog for:  opencpn-dashboard_pi
07:55:49.747 MESSAGE OCPNPlatform.cpp:1044 Loaded translation catalog for:  opencpn-wmm_pi
07:55:49.747 MESSAGE OCPNPlatform.cpp:1044 Loaded translation catalog for:  opencpn-chartdldr_pi
07:55:49.747 MESSAGE OCPNPlatform.cpp:1049 Loading catalog for opencpn core.
07:55:49.747 MESSAGE OCPNPlatform.cpp:890 Getting DefaultSystemLocale...
```
Do you have further suggest?
Thank you

keesverruijt, 2023-07-11:
I have suggested the wrong wx version for you as you are using Ubuntu 20.04 with an O version that is compiled for wx 3.0:

07:55:24.899 MESSAGE ocpn_app.cpp:1231 wxWidgets version: wxWidgets 3.0.4 Linux 64 bit wxGTK
...
07:55:24.902 MESSAGE ocpn_app.cpp:1250 OCPN_OSDetail:  x86_64 ; Ubuntu ; 20.04 ; ubuntu ; debian

The entire wx 3.0 / 3.2 question is a big mess: OpenCPN really needs the functionality of 3.2 but on older Linux releases wx 3.0 is still the standard version. This is why they recommend flatpak.

Is there a reason you want to compile this yourself?

You can save yourself a whole heap of trouble by compiling inside docker containers, so that you can restart from scratch more easily. Would be cool if our compiling instructions helped you more with this. Once we had CI working we haven't invested much in this.

closed 2026-02-11 by keesverruijt