opencpn-radar-pi/radar_pi #237: Radar_pi keeps initializing in Debian

agojc, 2023-05-25:
Hi there!
Plugin works fine in Win10 with a Halo radar after firewall was set up.
On the same PC, with other drives and a Debian 11 (bullseye) and derived-Debian 11 distribution with Opencpn 5.6.2 or 5.8.2 and Radar plugin 5.5.0, PPI keeps complaining "No radar/Initializing" and the wired network icon in the panel (taskbar) keeps chasing its tail.
Only with 4G or BR24 (which obviously can't figure out the serial), PPI shows it is scanning the PC's IP.
Couldn't find a solution to this or any sort of help on setting the right parameters if that's the issue.
Any help would be appreciated.
Thanx.
Jc

keesverruijt, 2023-05-25:
I think it is significant that your network connection doesn't finish setting up. Try giving it an explicit fixed IP in the 192.168/16 range.

The network card must have an IPv4 address for the radar plugin to be able to use it.

agojc, 2023-05-25:
Thanx for the reply.
Yes, i thought so.
Already tinkered a little bit with the parameters but couldn't figure out the right way to do it.

IPv4 is in DHCP mode. Should i set it in manual (i know from when i tried BR24 that it's listening on address 192.168.1.11, the local PC's IP), and what about netmask, gateway, and DNS servers...?

agojc, 2023-05-25:
Having attempted various settings, it seems there's some kind of conflict between ethernet and wifi.
It's been working at some point when i set both connections up in manual mode with fixed IPs.
But still very unstable... This is gonna need a little bit more digging.
Anyways, thanks for the help.
I'll come back here to post whatever useful information i get in the process, if any!

agojc, 2023-05-31:
So... I'm back as promised. Actually, i didn't look any further into my network/PC issues since it was intended as a test for my missing Raspberry Pi 4 with Openplotter installed.
Having been able to put my hands on a recently bought Pi4 (spare-to-be for the original one ), i sloted in the SDcard copy of the said system i keep, and it worked pretty much instanttly.
Radar plugin doesn't even complain with the wifi connection!
Didn't quite answer the initial question though, i guess...

Hakansv, 2023-05-31:
> Radar plugin doesn't even complain with the wifi connection!

No he should not but the function may be heavily disturbed dependent on the function of the wifi-router. 
All "blobs" (echoes) may not be transferred.

agojc, 2023-05-31:
So it's better off, you mean?
Knowing that wifi is used to connect to the internet in port (where there is no use of a radar, obviously!) and to display OpenCPN via VNC on a tablet (headless Pi4) at sea.
The Halo uses the wired connection only.
Do you think that this is usable, or not?

keesverruijt, 2023-05-31:
Yes having WiFi enabled is fine as long as the halo is on the wired connection and the network range does not overlap. So not both 192.168.x.x or both 10.1.1.x (as examples.)

agojc, 2023-05-31:
Thanks for this detailed information.

closed 2023-07-11 by keesverruijt