opencpn-radar-pi/radar_pi #238: Dopler Halo 20

fneveu29, 2023-06-30:
I have connecting a halo [Radar](https://www.cruisersforum.com/forums/tags/radar.html) with [opencpn](https://www.cruisersforum.com/forums/tags/opencpn.html) 5.8 and the [plug](https://www.cruisersforum.com/forums/tags/plug.html) in [Radar](https://www.cruisersforum.com/forums/tags/radar.html) 5.5. When I'm starting the Doppler mode, it 's start without problem but stop after 3 or 5 seconds.

keesverruijt, 2023-06-30:
Note that Doppler is not supported on HALO 20; it is the equivalent of the older "3G" radar so it only has a single radar, no B radar, lower power, fewer features.

We will try to use the .pcap data to determine if we can check whether a radar is a HALO 20 or 20+.

fneveu29, 2023-06-30:
here is the file .zip. I haven't thought to zip file, sorry.
[HALO 20.zip](https://github.com/opencpn-radar-pi/radar_pi/files/11916364/HALO.20.zip)
Thank you

keesverruijt, 2023-06-30:
If someone else has a PCAP file from a HALO20+ that would be helpful.

keesverruijt, 2023-06-30:
Could you upload another file where the radar is NOT transmitting and you power it up?

- Set radar to standby
- Switch radar off
- Stop opencpn
- Start wireshark / pcap recording
- Start opencpn
- Switch radar on
- Wait for radar to be detected, do not switch it on, and give it a command like change gain or range.
- Stop wireshark.
- Zip and upload.

Thanks!

susteranna, 2023-07-02:
As requested.
[halo20+.pcap.gz](https://github.com/opencpn-radar-pi/radar_pi/files/11929215/halo20%2B.pcap.gz)

fneveu29, 2023-07-23:
Hello, 
Is it possible to upload old versions of Radar_pi (5.2 or 5.3 from spring 2022)? I haven't find it anywhere.

keesverruijt, 2023-07-23:
Please do not ask multiple unrelated things in a single issue. This is not a forum but a code repository bug tracker.

fneveu29, 2023-07-23:
Sorry...

fneveu29, 2023-07-28:
yesterday I reinstalled Open cpn version 5.6.0 and radar pi  version 5.3.2 , as you can see in the attached picture the doppler function works perfectly without stopping
[image]

keesverruijt, 2023-07-28:
No it doesn't. It _thinks_ it enables Doppler, and it is not picking up the fact that the radar is reporting that Doppler isn't working. It is then using value 15 in every reported nibble as "approaching", which is why it is drawing all those stationary targets around you (the entire marina) in yellow. In fact this confirms that Doppler is NOT working in HALO20!

This was (silently) fixed in commit 80d646777 where we are now reading reports with length 22.

fneveu29, 2023-09-04:
New try out of the harbour yesterday with open cpn 5.6.0 and radar pi 5.3.2.
Doppler mode on, we can see each target identified by the radar and the movement of each. For me it's a good help to track other boats
[image]