opencpn-radar-pi/radar_pi #239: Menu buttons text are disturbed when O's Dark/dusk color scheme have been used.

Hakansv, 2023-07-07:
Compare [this CF post](https://www.cruisersforum.com/forums/f134/radar-plugin-v5-0-0-released-216057-52.html#post3793325)

When dark/dusk color scheme are or have been used menu buttons text are not correct shifted. 
One example using Emulator:
"Threshold" is the first button in the "Advanced" menu. That text is used for a lot other buttons in dusk color scheme.
Although a false button text the underlying function is correct.
The one method to come back to correct text function after a non daylight scheme have been used is to restart OCPN.

Examples...
Menu: "Advanced=>Installation"
[image]
Menu: "View"
[image]

keesverruijt, 2023-07-11:
This is probably a Windows only issue. This suggests it is a bug in wxWidgets as the other platforms don't have this problem as far as I can see.

As I don't use Windows regularly this may prove to be hard to fix.

@Hakansv can you provide an EXACT set of actions how you get this problem? E.g. with ALL button presses? Is it OK if the Menu dialog stays open during the switch to Dusk/Night mode?

Hakansv, 2023-07-11:
I'll try to fill more details:
The repeated menu text seems to always be the first used in the "Advanced box" described in ConstrolsDialog.cpp: 695
For a 4G where "Noise rejection" is active, as for Franziska, that text is reused. For me at home using Emulator, where "Noise rejection" is not use, the next used menu text is "Threshold" and here that's the one being reused.

> Is it OK if the Menu dialog stays open during the switch to Dusk/Night mode?

No. The failure is there as soon a color scheme, dusk/night, is present. No matter if I "Hide radar" or show it. Also if disable radar_pi, restart OCPN, set it to dusk mode and enable radar_pi. So it's not about a certain set of actions to force the failure. It's there all time a dusk/night color scheme is used and all time after that. Even when the color scheme is again reset to day mode. A O restart is needed to get rid of the failure.

As said it's the first used menu text in the "Advanced" menu that's used. In my case with Emulator it's "Threshold".
But the text "Threshold" is not used everywhere. Here comes a collection of all affected menu texts:
Button Adjust: (Should have shown "Range")
[image]
Button Advanced=>Installation:
[image]
Button View:
[image]
Button Guard zones (Is OK)
[image]
Button Window. (The second last should be "Transparency")
[image]

I could try to debug this but I'm not sure where set a brake and what details to look for. Any advises?

Hakansv, 2023-07-12:
FWIW: 
- The buttons affected are only those using wxWindowID: ID_CONTROL_BUTTON.
- When O dusk/night mode is active also the background color for our buttons are changed to gray. I've not so far been to figure out where this background color change is coded.